adafruit/circuitpython #4036: Implement busio.UART

tannewt, 2021-01-21:


matemaciek, 2021-02-11:
I want to connect to pico's REPL in another way than USB, as I want to use USB as host. Is it possible now, or should I wait for this issue to be done first?

tannewt, 2021-02-11:
@matemaciek We don't currently support REPL over UART or USB Host. We really want folks to *always* be able to access their code over USB via mass storage.

kmatch98, 2021-02-12:
@matemaciek I have heard that the CircuitPython crew has UART on the roadmap for the Pico but there is none on the map for USB host support (on any board).  

But once UART is available you can use a separate board as USB host to connect to the PICO via UART or other protocol. It’s not so elegant but you can “spoof” a REPL to go to UART. I know It’s not officially supported by the crew here but you can hack something that can work. Ping me on discord if you want an example.

dhalbert, 2021-02-12:
We are working on adding a second USB CDC serial channel, which will not be connected to the REPL, and could be used to communicate with a host to send an arbitrary binary bytestream back and forth.

alexbartlow, 2021-02-15:
👍 for this feature.

Since the pico has two UART ports, I was hoping to use one of them to talk to another pico - daisy chaining them together would be great, with child picos using UART0 to talk to the parent, and UART1 to talk to the next child in the chain.

As it is now, the pico does not support running as an I2C peripheral under circuitpython, and can not use the UART1 function, so the options are limited for pico-to-pico communication.

tsukisan, 2021-02-17:
+1

I'm looking to connect a serial LCD and GPS for telemetry and speed in a vehicle.

ebreyit, 2021-02-18:
👍 for this feature.
Serial comms for remote Vision Mixer and Camera control

ladyada, 2021-02-19:
hi folks please try
https://github.com/adafruit/circuitpython/suites/2077728823/artifacts/42065165
from PR https://github.com/adafruit/circuitpython/pull/4224
readthedocs: https://circuitpython.readthedocs.io/en/latest/shared-bindings/busio/#busio.UART

tylercrumpton, 2021-02-19:
> hi folks please try
I haven't done anything exhaustive, but I was able to verify some of the functionality of the UART0 communication with a project I had set up already! Thanks for putting this up!

closed 2021-02-26 by tannewt