adafruit/circuitpython #4045: func .__ code__ ?

lwerner-lshigh, 2021-01-22:
I know this was asked for in micropython 5ish years ago, but it seems that it still hasnt been implemented. I was wondering if this would be possible to add? Im currently working on creating a calculator that can be plugged in and have programs modified. The menu system ended up using that functionality. 
```python
class Program(object):

    def exec():
        '''to be implemented in programs'''
        pass

    def run(self):
        params = self.exec.__code__.co_varnames[1:self.exec.__code__.co_argcount]
        argList = []
        for param in params:
            argList.append(float(input(f"{param}: ")))
        funcKwarg = dict(zip(params, argList))
        print(f"Result: {self.exec(**funcKwarg)}")
```
If this isnt possible thats also fine, just wondering if it is possible.

tannewt, 2021-01-22:
Is this an existing CPython API? I've never heard of it.

Is there a corresponding MicroPython issue?

Please elaborate on how you are using it. There might be other ways to do it.

lwerner-lshigh, 2021-01-22:
I believe this is a CPython API as it is in the python 3 standard types [here](https://docs.python.org/3/library/stdtypes.html?highlight=__code__#code-objects). This API allows for inspection/introspection of python functions. Here is a test program from [python/cpython](https://github.com/python/cpython/blob/master/Lib/test/test_code.py). 
The corresponding MicroPython issue is [#2084](https://github.com/micropython/micropython/issues/2084l)

I am using `__code__` to be able to inspect the parameters of a function defined in another file. The calculator will be able to select programs in the programs.py file which will have its math function defined in `exec()`

tannewt, 2021-01-25:
Thanks for the link to the MP issue. I'd take Damien's word that it's not an API we should add.

It seems weird to me that you want to show info about a python function in a menu. I'd expect separate metadata for "plugins" or to simply call the new function for python code that imports it.

lwerner-lshigh, 2021-02-04:
I think i have a solution... not as elegant but ill just provide the "metadata" by hand as a property.
```python
class Program(object):

    '''to be filled out in programs to let us know the input types they want'''
    props = {}

    def exec():
        '''to be implemented in programs'''
        pass

    def run(self):
        inputParams = {}
        for param in self.props:
            inputParams[param] = self.props[param](
                    input(f"{param}: ")
            )
        print(f"Result: {self.exec(**inputParams)}")
```
and here is a sample implementation of this class 
```python
class Vertex(Program):
    '''Find the vertex of a parabala'''

    props = {
        'a': float,
        'b': float,
        'c': float
    }

    def exec(_, a,b,c):
        x = ((-b)/(2*a))
        return x, ((a * (x)**2) + (b * (x)) + c)
```
repl output...
```python
Adafruit CircuitPython 6.1.0-beta.2-204-gf9d9c031a on 2020-12-28; FeatherS2 with ESP32S2
>>> import programs
>>> programs.Vertex().run()
c: -12
a: -1
b: -8
Result: (-4.0, 4.0)
>>>

```

closed 2021-02-04 by lwerner-lshigh