adafruit/circuitpython #4057: ESP32-S2 "Out of sockets" in 6.2.0-beta.0

anecdata, 2021-01-24:
Context manager __exit__ seems to be in place for socket, but:
```
import time
import wifi
import socketpool

from secrets import secrets

pool = socketpool.SocketPool(wifi.radio)

print("Connect...")
wifi.radio.connect(secrets["ssid"], secrets["password"])
print("RSSI", wifi.radio.ap_info.rssi)

count = 0
packet = bytearray(1)
while True:
    count += 1
    print(count, end=" ")
    packet[0] = 0xFF
    with pool.socket(pool.AF_INET, pool.SOCK_DGRAM) as sock:
        print("Send...")
        sock.sendto(packet, ("192.168.4.32", 16384))
    time.sleep(1)
```
results in:
```
Adafruit CircuitPython 6.2.0-beta.0 on 2021-01-22; FeatherS2 with ESP32S2
>>> import ntp
Connect...
RSSI -43
1 Send...
2 Send...
3 Send...
4 Send...
5 Send...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ntp.py", line 21, in <module>
  File "ntp.py", line 21, in <module>
RuntimeError: Out of sockets
```
CPython version continues past where my patience wears out:
```
#CPython version
import time
import socket

pool = socket

count = 0
packet = bytearray(1)
while True:
    count += 1
    print(count, end=" ")
    packet[0] = 0xFF
    with pool.socket(pool.AF_INET, pool.SOCK_DGRAM) as sock:
        print("Send...")
        sock.sendto(packet, ("192.168.4.32", 16384))
    time.sleep(1)
```
May be related to [https://github.com/adafruit/Adafruit_CircuitPython_Requests/issues/63](https://github.com/adafruit/Adafruit_CircuitPython_Requests/issues/63)

hierophect, 2021-01-25:
Probably unrelated to the other issue, other than the general problem of "Hierophect is bad at sockets", ha ha. There's supposed to be a subsystem that tracks what sockets have been initialized and makes sure they shut down on reset or close, but that doesn't seem to be working here. I'll give it another spin with your code and see if I can pin it down.

tannewt, 2021-01-25:
@hierophect I don't think it's your problem. I probably never made exit call close. (It should automatically.)

hierophect, 2021-01-25:
@tannewt no, it's my bad, I forgot to make Close de-register handles from the tracking array, only `reset_sockets` was doing it. I can fold the change into the SSLSocket PR.

closed 2021-02-02 by tannewt

KTibow, 2021-02-15:
I'm still getting this issue with `magtag.network.fetch` on 6.2.0-beta.2.

anecdata, 2021-02-15:
@KTibow What version of `adafruit_requests` do you have? There were a couple of soket leaks plugged recently. Can you try the latest if you're not already using it?

jfabernathy, 2021-02-18:
I posted a question in the forum and they directed me here:
I just updated my UF2 file to 6.2.0 Beta 2 and tried the Internet Test from the Learn Guide
https://learn.adafruit.com/adafruit-metro-esp32-s2/circuitpython-internet-test
I get the out of sockets error running the Internet test:
`Fetching and parsing json from https://api.github.com/repos/adafruit/circuitpython
Traceback (most recent call last):
  File "code.py", line 55, in <module>
  File "adafruit_requests.py", line 598, in get
  File "adafruit_requests.py", line 559, in request
  File "adafruit_requests.py", line 436, in _get_socket
  File "adafruit_requests.py", line 433, in _get_socket
RuntimeError: Out of sockets`
The Error occurs on the second time the code runs. BTW I also updated the Libraries today as well.
[code.txt](https://github.com/adafruit/circuitpython/files/5999431/code.txt)

jfabernathy, 2021-02-18:
Never mind. I got pointed the the latest version that has not been release yet, but is in the pipeline. It worked._

PR #4049 (2021-02-02): Separate SSLSocket from Socket
This PR removes the TLS implementations from Socket and moves them to a new object class, SSLSocket, which is created from a Socket object using SSL's `contex.wrap_socket()` function. This behavior more closely matches that of [Cpython](https://docs.python.org/3/library/ssl.html#ssl-sockets).

Opening as a draft as I've had a couple difficulties with this change and I'm still testing:

- The regular Socket class has a new LWIP-only implementation for `connect`, but I've had issues making this non-blocking. A non-blocking call to `connect` should always return a - [... cut]