adafruit/circuitpython #4058: Compiling mpy-cross fails

heke-pp, 2021-01-25:
I am attempting to compile the mpy-cross on Linux Mint 19 Cinnamon. Python version 3.6.9. No git installed.
With make getting error:
  File "../py/makeversionhdr.py", line 72, in make_version_header
    git_tag, git_hash, ver = info
TypeError: 'NoneType' object is not iterable

In ../py/makeversionhdr.py
Seems that on line 70:
        info = get_version_info_from_docs_conf()
can return None, which then raises error on 72.

Line 57:
           if line.startswith("version = release = '"):
Has an extra ' before the closing ".
The line to search in conf.py is : "version = release = final_version"

After removing the ', make run successfully 
(well...until getting error "ModuleNotFoundError: No module named 'huffman'", but that's a separate issue)

tannewt, 2021-01-25:
Why not just install `git`?

dhalbert, 2021-03-31:
Closing due to no activity. Please re-open or ask for support via https://adafru.it/discord and https://forums.adafruit.com if you are still having trouble.