adafruit/circuitpython #4059: .local domains returns OSError: 0

wifijt, 2021-01-25:
I was trying to use the home assistant API and I ran into an issue using the homeassistant.local domain in the fetch request. This caused an OSError 0 

Retrieving data...Traceback (most recent call last):
  File "code.py", line 30, in <module>
  File "code.py", line 27, in <module>
  File "adafruit_magtag/magtag.py", line 174, in fetch
  File "adafruit_portalbase/__init__.py", line 365, in fetch
  File "adafruit_portalbase/network.py", line 523, in fetch_data
  File "adafruit_portalbase/network.py", line 442, in fetch
  File "adafruit_requests.py", line 595, in get
  File "adafruit_requests.py", line 559, in request
  File "adafruit_requests.py", line 419, in _get_socket
OSError: 0

The workaround is to use the IP address of the server.

anecdata, 2021-01-25:
`.local` domains seem to resolve in general:
```
Adafruit CircuitPython 6.1.0-rc.1 on 2021-01-19; FeatherS2 with ESP32S2
>>> import wifi
>>> import socketpool
>>> from secrets import secrets
>>>
>>> pool = socketpool.SocketPool(wifi.radio)
>>> wifi.radio.connect(secrets["ssid"], secrets["password"])
>>> wifi.radio.ap_info.rssi
-51
>>> wifi.radio.ipv4_dns
1.1.1.2
>>> pool.getaddrinfo("desktop.local", 443)
[(0, 0, 0, '', ('192.168.1.32', 443))]
```
One possibility is a LAN DNS proxy or something else interfering with the name resolution.

wifijt, 2021-01-25:
Hmm it's something about the homeassistant.local mdns query because other .local domains resolve

>>> pool.getaddrinfo("wind.local",443)
[(0, 0, 0, '', ('192.168.1.201', 443))]
>>> pool.getaddrinfo("homeassistant.local",443)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 0
>>>

tannewt, 2021-01-25:
I'm not sure we have mDNS enabled.

dhalbert, 2022-05-14:
@tannewt This bug is fixed by the new mDNS code?