adafruit/circuitpython #4067: Hook up RTC for sleeping and timekeeping

tannewt, 2021-01-26:


tannewt, 2021-02-02:
https://github.com/adafruit/circuitpython/pull/4110 Allows us to set the RTC. For sleeping, we need to modify https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/supervisor/port.c#L164-L187 for sleep support.

xorbit, 2021-02-05:
I don't know enough to know if it's related to this issue, but where `time.time()` used to fail because no RTC was attached, it now seems to produce seemingly random numbers (but they're from a repeating set) on the RPi Pico.

tannewt, 2021-02-05:
@microdev1 any idea if your RTC work should have fixed time.time?

microDev1, 2021-02-12:
It seems like [`common_hal_rtc_init()`](https://github.com/adafruit/circuitpython/blob/08f30fedf5401dd6aa9bf318e4d1eda59ae5652f/ports/raspberrypi/common-hal/rtc/RTC.c#L33-L35) is not getting called which leads to generation of random numbers. 
I am not sure where to add a call for it, how do other ports handle RTC initialization?

tannewt, 2021-02-12:
@microDev1 I suspect RTC init is done in `port_init` in `supervisor/port.c`.n

microDev1, 2021-02-16:
`port_init` works... since this is a small change I might include it with my `UART` work or maybe you can fit it in somewhere. :)

tannewt, 2021-07-15:
Fixed by #4816

closed 2021-07-15 by tannewt