adafruit/circuitpython #4074: Porting APDS-9930 library

ilium007, 2021-01-27:
I would like to look at porting the APDS-9930 python library to circuit python. I am building some boards for an automation project that will use the APDS-9930, I don't need RGB sensing and the 9930 is more that 2 times cheaper than the 9960. I am hoping I can use the 9960 library as a framework. Do "non Adafruit" hardware libraries ever get pulled into this repo? I am looking at using circuit python on the the main control uC's but wont be using Adafruit specific boards for sensors.

ilium007, 2021-01-27:
Python APDS-9930 here:
https://github.com/Depau/python-apds9930

jepler, 2021-01-27:
Have a look at our guide [Creating and Sharing a CircuitPython Library](https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/overview).  We'd love to see your library contributed to the CircuitPython "Community Bundle", which includes drivers written by 3rd parties for hardware Adafruit doesn't sell or support.

Consider dropping by [our Discord](https://adafru.it/discord) or using the [Adafruit Forums](https://forums.adafruit.com/index.php?) if you run into trouble. And thanks for offering to help the community by developing a driver!

closed 2021-01-27 by jepler