adafruit/circuitpython #4079: ESP32-S2: No I2C device at address error after reset

makermelissa, 2021-01-27:
It works from a cold boot, but after a Control-D reset, it doesn't find a device anymore until the next cold reboot. Tested with the ESP32-S2 Metro and SHTC3 using absolute Newest CircuitPython.

makermelissa, 2021-01-27:
I just tried this with the new 6.2 Beta 1 and the issue appears to be gone.

makermelissa, 2021-01-29:
I was able to recreate this in 6.2.0.beta-1. It happens when you use board.I2C(), but appears to function fine if you use busio.I2C(board.SCL, board.SDA).

makermelissa, 2021-01-29:
It also continues misbehaving even after switching to the busio method until you hard reset.

makermelissa, 2021-01-29:
I can't get this to work at all anymore. Even after a deep sleep it fails.

ThomasAtBBTF, 2021-01-29:
This seems to be a similar problem as described in https://github.com/adafruit/circuitpython/issues/4046

hierophect, 2021-02-15:
I'm consolidating this with a number of other issues that all appear to be the same thing. Please continue discussion of this issue in #4046.

closed 2021-02-15 by hierophect

richsad, 2022-02-28:
FWIW, I am getting this error today using 7.2.0-alpha.2 on 2022-02-11; Adafruit QT Py ESP32S2 with ESP32S2 using board.STEMMA_I2C() with FRAM plugged in. Error is:
`code.py output:
Traceback (most recent call last):
  File "code.py", line 17, in <module>
  File "adafruit_fram.py", line 216, in __init__
ValueError: No I2C device at address: 0x7c
`

dhalbert, 2022-02-28:
Please try 7.2.0 final.

dhalbert, 2022-02-28:
@richsad Please open a new issue, because your error is different, and also please try with 7.2.0 final.