adafruit/circuitpython #4081: Reset time.monotonic every VM reset

tannewt, 2021-01-27:
To do this, we can store the internal tick time that the vm starts up and subtract it from the current time before returning monotonic. This way the resolution will reset with the VM and also be agnostic to depending on the RTC value.

sixtyfive, 2021-04-25:
Hi, is this something that someone with Arduino-level knowledge of C++ (but no direct register manipulation and such) could conceivably write as  a PR? And if so, could you provide some guidance as to what would need to be done?

gamblor21, 2021-04-25:
I came into doing core circuitpython code with a similar knowledge level. I think it is definitely possible to do. Have you set up a build environment to build the core code yet? Once I got past that step, I just started going through the code to learn what was there. I'm not 100% sure where the VM reset is but the discord channel `circuitpython-dev` is always able to provide guidance as well.

sixtyfive, 2021-04-25:
Yes, I've been building my own `.uf2` and `.mpy` files for a while now (mostly because I needed to modify the sources for the latter - never touched the sources for the the former. Right now, I'm grepping around for where `supervisor.reload()` might be defined but not finding it (only thing related to _auto_-reload)....

dhalbert, 2021-04-25:
Look in `shared-binding/supervisor/__init__.c` for `supervisor_reload()`, and then see what that calls. (I'm leaving that as an exercise so you can discover the structure :slightly_smiling_face:). In general, you can look for `MP_QSTR_some_name`, to see where `some_name` that is a Python name is used.

dhalbert, 2021-04-25:
Also see https://circuitpython.readthedocs.io/en/latest/docs/index.html#project-structure and these links in the left sidebar in ReadTheDocs:
[image]
There is also this guide, but there is much about it that is now outdated: https://learn.adafruit.com/extending-circuitpython

tannewt, 2021-04-26:
Here is where the tick times are returned: https://github.com/adafruit/circuitpython/blob/main/shared-module/time/__init__.c This is where I'd subtract the VM start up time from the current time. I don't actually want the internal time to change.

tannewt, 2021-07-15:
Use `supervisor.ticks_ms()` as added by #4936

closed 2021-07-15 by tannewt