adafruit/circuitpython #4085: ESP32S2 TCP Server: Error EAGAIN on socket.accept()

Larrxi, 2021-01-28:
I try to start a TCP server on my Lilygo TTGO T8 ESP32-S2 with display.

```
code.py output:
192.168.43.83
will timeoutTraceback (most recent call last):
  File "code.py", line 18, in <module>
OSError: [Errno 11] EAGAIN

Code done running.
```

I am using the code from here:
https://gist.github.com/hierophect/5bb6133a69f8395841531013d46d2703

hierophect, 2021-01-29:
I think this is probably because the Server Socket PR inadvertently changed the default timeout of sockets to 0 instead of None. You could try increasing the timeout with `settimeout(None)` or try #4049 which will revert to a no-timeout default.

Larrxi, 2021-01-29:
With settimeout -1 it give other error EATIMEOUT or something and it did print out the client IP address.

I then try with #4049 (without settimeout) and now it proceeds further:

```
code.py output:
192.168.43.83
Connected by ('192.168.43.17', 22723)
bytearray(b'Hello, world\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
```

The client gets:
`Received b''`

Larrxi, 2021-02-01:
Okey it works when, before connecting socket, setting socket.settimeout(None).
Thanks!

hierophect, 2021-02-01:
@Larrxi no problem. Note that since #4095 has now been merged, you shouldn't even need to take that step anymore on the most recent builds, as the default behavior has been returned to a timeout of None.

closed 2021-02-01 by hierophect