adafruit/circuitpython #4086: ulab: unexpected result of ulab.numerical.std(list)

bspsbee, 2021-01-28:
Running the benchmark code from [A Simple Benchmark](https://learn.adafruit.com/ulab-crunch-numbers-fast-with-circuitpython/a-simple-benchmark)

From `adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0-rc.0.uf2` forward, the result of `ulab.numerical.std(nums_list)` is larger than expected. 35358 instead of 3535.

First observed on Raspberry Pi Pico.

Tested release versions on ItsyBitsy M4:
```
adafruit-circuitpython-itsybitsy_m4_express-en_US-6.2.0-beta.1.uf2
traditional                    :    2.228ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.284ms [result=3535.853624]
ulab only, with list           :    0.159ms [result=35358.531475]
ulab only, with ndarray        :    0.072ms [result=3535.853624]

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.2.0-beta.0.uf2
traditional                    :    2.390ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.288ms [result=3535.853624]
ulab only, with list           :    0.157ms [result=35358.531475]
ulab only, with ndarray        :    0.072ms [result=3535.853624]

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0.uf2
traditional                    :    2.318ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.283ms [result=3535.853624]
ulab only, with list           :    0.159ms [result=35358.531475]
ulab only, with ndarray        :    0.072ms [result=3535.853624]

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0-rc.1.uf2
traditional                    :    2.386ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.282ms [result=3535.853624]
ulab only, with list           :    0.157ms [result=35358.531475]
ulab only, with ndarray        :    0.071ms [result=3535.853624]

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0-rc.0.uf2
traditional                    :    2.316ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.289ms [result=3535.853624]
ulab only, with list           :    0.159ms [result=35358.531475]
ulab only, with ndarray        :    0.071ms [result=3535.853624]

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0-beta.3.uf2
ImportError: no module named 'ulab.numerical'

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0-beta.2.uf2
ImportError: no module named 'ulab.numerical'

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0-beta.1.uf2
traditional                    :    2.258ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.231ms [result=3535.853624]
ulab only, with list           :    0.262ms [result=3535.854340]
ulab only, with ndarray        :    0.061ms [result=3535.854340]

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.1.0-beta.0.uf2
traditional                    :    2.305ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.231ms [result=3535.853624]
ulab only, with list           :    0.260ms [result=3535.854340]
ulab only, with ndarray        :    0.060ms [result=3535.854340]

adafruit-circuitpython-itsybitsy_m4_express-en_US-6.0.1.uf2
traditional                    :    2.267ms [result=3535.843611]
ulab, with ndarray, some implementation in python :    0.229ms [result=3535.853624]
ulab only, with list           :    0.261ms [result=3535.854340]
ulab only, with ndarray        :    0.059ms [result=3535.854340]
```

jepler, 2021-01-28:
@v923z can you take a look?

v923z, 2021-01-29:
@bspsbee Thanks for bringing up the issue and the detailed report! 

@jepler It seems to me that this crept in between 6.1.0-beta.1, and 6.1.0-rc.0. Can you point me to the `ulab` commit tags that you link to for these two versions? You found a small glitch with the `std` function a while ago, and I am pretty sure that I messed up something, when I tried to iron that out.

jepler, 2021-01-29:
I think the refs are ... 
```
jepler@eric:~/src/circuitpython$ git diff 6.1.0-beta.1 6.1.0-rc.0 -- extmod/ulab/
diff --git a/extmod/ulab b/extmod/ulab
index 8242b8475..c4b06e419 160000
--- a/extmod/ulab
+++ b/extmod/ulab
@@ -1 +1 @@
-Subproject commit 8242b84753355433b61230ab6631c06e5ac77f35
+Subproject commit c4b06e419f3d515478b05bb8ce03ebdb29cddec4
```
there are a lot of ulab commits in that range, because CircuitPython had fallen behind.
```
$ git rev-list 8242b84753355433b61230ab6631c06e5ac77f35..c4b06e419f3d515478b05bb8ce03ebdb29cddec4 | wc -l
280
```

v923z, 2021-01-29:
@jepler Thanks! I have found the glitch, the fix is on the way. I'll ping you, once it is ready.

jepler, 2021-01-29:
Excellent, thank you!  I'll pull-request the updated legacy branch into our main branch.

PR #4089 (2021-01-29): ulab: update to 1.7.2
Closes #4086

Thanks to @v923z for a prompt fix to the underlying problem!

@bspsbee can you let us know if this change fixes your problem?
commit 689fba76 (2021-01-29): ulab: update to 1.7.2

Closes #4086
closed 2021-01-29 by jepler