adafruit/circuitpython #4091: Missing sys.platform on raspberrypi ports

jamesbowman, 2021-01-29:
``sys.platform`` is missing from the raspberrypi ports.

I started to make a PR, with ``#define MICROPY_PY_SYS_PLATFORM`` in ``mpconfigport.h``. However, it seems to require some thought. According to the docs for ``sys.platform``:

> For baremetal ports it is an identifier of the chip on a board, e.g. "MicroChip SAMD51". It thus can be used to distinguish one board from another.

(This is exactly what I want it for... to distinguish between boards in the Gameduino driver and assign the appropriate default IO pins.)

So it should be different for the raspberry_pi_pico and adafruit_feather_rp2040 boards. Maybe it should be

    "Rpi RP2040"
    "Adafruit RP2040"

But unsure. Happy to continue the PR with any given scheme.

dhalbert, 2021-01-29:
`sys.platform` usually just indicates the chip that's used, not the individual board. That documentation could be phrased better. Take a look at `sys.uname()`, which provides board info:
```python
>>> os.uname()
(sysname='samd21', nodename='samd21', release='6.2.0', version='6.2.0-beta.1-9-g70a5f464d on 2021-01-29', machine='Adafruit Trinket M0 with samd21e18')
```

dhalbert, 2021-01-29:
`os.uname()` on a Pico:
```python
>>> os.uname()
(sysname='rp2040', nodename='rp2040', release='6.2.0', version='6.2.0-beta.1 on 2021-01-27', machine='Raspberry Pi Pico with rp2040')
```

jamesbowman, 2021-01-29:
Excellent, thanks.

So should I send a PR with:

#define MICROPY_PY_SYS_PLATFORM "rp2040"

for the raspberrypi ports?

dhalbert, 2021-01-29:
Yes, that would be good, in `mpconfigport.h`

dhalbert, 2021-01-29:
Use `RP2040`, capitalized, because that's the chip designation.

dhalbert, 2021-01-30:
Fixed by #4092

closed 2021-01-30 by dhalbert