adafruit/circuitpython #4093: ConnectionError when attempting to connect to a known SSID (6.2.0b1)

brentru, 2021-01-29:
wifi.radio.connect() call seems to improperly raise a `ConnectionError: No network with that ssid` even when the SSID is found by a WiFi scan.

Platform: `Adafruit CircuitPython 6.2.0-beta.1 on 2021-01-27; Adafruit MagTag with ESP32S2`
Code
```
import ipaddress
import ssl
import wifi
import socketpool

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

print("ESP32-S2 WebClient Test")

print("My MAC addr:", [hex(i) for i in wifi.radio.mac_address])

print("Available WiFi networks:")
for network in wifi.radio.start_scanning_networks():
    print("\t%s\t\tRSSI: %d\tChannel: %d" % (str(network.ssid, "utf-8"),
            network.rssi, network.channel))
wifi.radio.stop_scanning_networks()

print("Connecting to %s"%secrets["ssid"])
wifi.radio.connect(secrets["ssid"], secrets["password"])
print("Connected to %s!"%secrets["ssid"])
print("My IP address is", wifi.radio.ipv4_address)
```

Output:
```
code.py output:
ESP32-S2 WebClient Test
My MAC addr: ['0x7c', '0xdf', '0xa1', '0x5', '0xfa', '0x26']
Available WiFi networks:
        NETWORKNAME         RSSI: -79       Channel: 1
....
Connecting to NETWORKNAME
Traceback (most recent call last):
  File "code.py", line 25, in <module>
ConnectionError: No network with that ssid

Code done running.
```

brentru, 2021-01-29:
I tried this on a Metro ESP32S2 running the 6.2.0b1 and it connected to my SSID. Might be  Magtag-specific

Platform: `Adafruit CircuitPython 6.2.0-beta.1 on 2021-01-27; Adafruit Metro ESP32S2 with ESP32S2`

anecdata, 2021-01-29:
Others have reported it on Discord. I've seen this on rare occasions on other esp32-s2 boards and prior releases, chalked it up to intermittent wifi issue since it usually recovers on subsequent retries. Does your exception persist?

brentru, 2021-01-29:
It does persist between retries and router restarts. Tried connecting to a new WiFi network I spun up on my phone, same issue.

jerryneedell, 2021-01-29:
@brentru Is this with NINA 1.7.1?
I do see occasional failed connects even if the SSID is in the scan, but it always recovers on retry.

jerryneedell, 2021-01-29:
Sorry -- this is on an S2  --- I'm not sure if I have seen the failed connects or not on S2.

anecdata, 2021-01-29:
`NO AP FOUND` is an "Espressif-specific Wi-Fi reason-code: when the station fails to scan the target AP, this reason code will be reported." there is the explicit scan you do in your code, but the esp32-s2 also scans implicitly when connecting, there could be a bug in the IDF. Just curious... does it also happen if you don't do an explicit scan before connecting (ruling out a bug in our stop_scanning_networks)?

jerryneedell, 2021-01-29:
BTW -- I just compiled from the tip of main 6.2.0-beta.1+ for metros2  and was able to connect ok
not seeing any failed connect/retry -- yet

anecdata, 2021-01-29:
I haven't seen this yet with >=`6.2.0-beta.1`, but then I'm <10' from the AP and RSSIs are in the -30s to -40s.

jerryneedell, 2021-01-29:
Also just ran it on a MagTag with out any problems ...

somenice, 2021-08-01:
Hi there Adafruit folks, 
I'm running into this similar error with the Funhouse which I think is built on a similar architecture.
https://learn.adafruit.com/creating-funhouse-projects-with-circuitpython/temperature-logger-example

`Traceback (most recent call last):
  File "code.py", line 69, in <module>
  File "code.py", line 59, in log_data
  File "adafruit_portalbase/network.py", line 345, in connect
  File "adafruit_portalbase/network.py", line 340, in connect
  File "adafruit_portalbase/wifi_esp32s2.py", line 62, in connect
ConnectionError: No network with that ssid

Code done running.

Adafruit CircuitPython 6.3.0 on 2021-06-01; Adafruit FunHouse with ESP32S2`

It runs successfully for a random amount of time before getting an error and multiple different access points were tested.

Any thoughts? Thank you.

dustin, 2022-06-15:
I also just ran into this issue.  Is it an RSSI problem?  I think it might be a single issue.  I've not looked into the code, but perhaps it rejects anything with a sufficiently low RSSI.

denschub, 2022-08-25:
So, I just have run into this issue on a MagTag, on both the latest CircuitPython 7 release, but also on the 8 Beta. Interestingly, this seems to only happen when the MagTag is plugged in via USB into my laptop. It's fine if it's running off the battery. Also, it works fine using the same USB cable on a powerbank.

So ultimately, a noisy power supply or a noisy ground level might jsut be to blame here...

OdorDecoder, 2022-09-09:
I have run into this issue today with my QT PY ESP32 S2. @denschub is right, about a noisy power source. I was using a 3' long USB-C cable, and then I switched to a 1' short USB-C cable and it connects to WiFi just fine. I was using the wifi connect example found here [https://learn.adafruit.com/adafruit-qt-py-esp32-s2/circuitpython-internet-test](url)

paulmand3l, 2022-09-26:
Running into this issue on a brand new Feather ESP32-S3 using https://learn.adafruit.com/adafruit-esp32-s3-feather/circuitpython-internet-test. Doesn't work when plugged into my computer's USB port but works on battery (and via 5V power from the [Matrix Featherwing](https://www.adafruit.com/product/3036)

bcransto, 2022-10-22:
I too cannot get my ESP32s to connect to wifi. I have two ESP32-S3s and one ESP32-S2 and all three are giving me "unknown failure" messages. I tried connecting while powering w lipo battery only. No luck. These boards were connecting to wifi a couple months ago but it was a different network. Any suggestions?

bcransto, 2022-10-23:
FYI... I just upgraded the circuit python on my boards. they're working now. 🤗

dhalbert, 2022-10-23:
Closing, given the age of this. Please reopen with an updated example if you still see this.

closed 2022-10-23 by dhalbert