adafruit/circuitpython #4094: Add optional pin pull.UP/DOWN to countio.Counter for open-collector sensors

CedarGroveStudios, 2021-01-29:
Would it be possible to add an optional pin _pull.UP_ attribute to _countio.Counter_? The capability would accommodate a variety of digital sensors with open-collector outputs, eliminating the need for an added pull-up resistor.

tannewt, 2021-02-01:
What are some example sensors? How much does the resistor value matter? It's usually better to be explicit with an external value.

CedarGroveStudios, 2021-02-01:
Before trying _countio_, an internal pull-up was sufficient to work with both optical and hall-effect interrupter sensors. The pull-up value is not critical for most low to medium-speed applications like motor RPM. Don't know if the internal pull-up value provides enough current for the speed of a remote control IR sensor.

Sensors tested:
Hall-Effect:  [https://www.adafruit.com/product/158](url)
IR Reflective: [https://www.adafruit.com/product/3930](url)
T-Slot Interrupter: [https://www.adafruit.com/product/3986](url)

Does _rotaryio_ provide pull-ups for encoder pins? I've been able to read encoders without pull-up resistors. Hmmm.

tannewt, 2021-02-02:
> Does _rotaryio_ provide pull-ups for encoder pins? I've been able to read encoders without pull-up resistors. Hmmm.

I don't think so. I think our guide has you hook one up.

tannewt, 2021-02-02:
> Before trying _countio_, an internal pull-up was sufficient to work with both optical and hall-effect interrupter sensors. The pull-up value is not critical for most low to medium-speed applications [...]

I'm ok if you want to add a kwarg for this.

CedarGroveStudios, 2021-02-02:
I'm completely unsure about where to start when it comes to modifying one of the core elements. Time to look at how that's done. Thanks.

ladyada, 2021-02-02:
@CedarGroveStudios you can do it! we can help support ya, try making a draft PR when you're ready for some eyes :)

CedarGroveStudios, 2021-02-02:
Thanks! Another fun trip outside my comfort zone!

@tannewt:  Interestingly, the _rotaryio_ guides I've looked through that involve CircuitPython have no pull-up resistors for the two quadrature encoder pins.  [https://learn.adafruit.com/rotary-encoder/overview](url)

tannewt, 2021-02-02:
Yup, I'm wrong. IncrementalEncoder on the SAMD turns them on automatically: https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/rotaryio/IncrementalEncoder.c#L59-L63

CedarGroveStudios, 2022-04-26:
According to the information in read the docs, pull up/down and None appear to be included now. I’ll test and confirm that it works with the listed sensors.

dhalbert, 2023-10-28:
Fixed by #5803.

closed 2023-10-28 by dhalbert