adafruit/circuitpython #4099: ld syntax error when trying to build for M4 with internal flash

gc-elec, 2021-01-30:
Hello,

I designed a custom board prototype based on ATSAMD51G19A and I try to build CircuitPython for it.
I can build without problem with external flash enabled, but I need a build with internal flash too.
When I try to build with INTERNAL_FLASH_FILESYSTEM = 1, I get an ld error :

```
arm-none-eabi-gcc -o build-0_m4/firmware.elf -DCIRCUITPY_FULL_BUILD=1 -DCIRCUITPY_AESIO=0 -DCIRCUITPY_ANALOGIO=1 -DCIRCUITPY_AUDIOBUSIO=0 -DCIRCUITPY_AUDIOIO=1 -DCIRCUITPY_AUDIOIO_COMPAT=1 -DCIRCUITPY_AUDIOPWMIO=0 -DCIRCUITPY_AUDIOCORE=1 -DCIRCUITPY_AUDIOMIXER=1 -DCIRCUITPY_AUDIOMP3=1 -DCIRCUITPY_BITBANGIO=0 -DCIRCUITPY_BLEIO=0 -DCIRCUITPY_BOARD=1 -DCIRCUITPY_BUSIO=1 -DCIRCUITPY_DIGITALIO=1 -DCIRCUITPY_COUNTIO=1 -DCIRCUITPY_DISPLAYIO=1 -DCIRCUITPY_FRAMEBUFFERIO=1 -DCIRCUITPY_VECTORIO=0 -DCIRCUITPY_FREQUENCYIO=0 -DCIRCUITPY_GAMEPAD=1 -DCIRCUITPY_GAMEPADSHIFT=0 -DCIRCUITPY_I2CSLAVE=0 -DCIRCUITPY_MATH=1 -DCIRCUITPY__EVE=0 -DCIRCUITPY_MICROCONTROLLER=1 -DCIRCUITPY_NEOPIXEL_WRITE=1 -DCIRCUITPY_NETWORK=1 -DCIRCUITPY_NVM=1 -DCIRCUITPY_OS=1 -DCIRCUITPY_PIXELBUF=1 -DCIRCUITPY_RGBMATRIX=1 -DCIRCUITPY_PULSEIO=1 -DCIRCUITPY_PS2IO=1 -DCIRCUITPY_RANDOM=1 -DCIRCUITPY_ROTARYIO=1 -DCIRCUITPY_RTC=1 -DCIRCUITPY_SAMD=1 -DCIRCUITPY_STAGE=0 -DCIRCUITPY_STORAGE=1 -DCIRCUITPY_STRUCT=1 -DCIRCUITPY_SUPERVISOR=1 -DCIRCUITPY_TIME=1 -DCIRCUITPY_TOUCHIO_USE_NATIVE=0 -DCIRCUITPY_TOUCHIO=1 -DCIRCUITPY_UHEAP=0 -DCIRCUITPY_USB_HID=1 -DCIRCUITPY_USB_MIDI=1 -DCIRCUITPY_PEW=0 -DCIRCUITPY_USTACK=0 -DCIRCUITPY_BITBANG_APA102=0 -DCIRCUITPY_REQUIRE_I2C_PULLUPS=1 -DCIRCUITPY_SERIAL_BLE=0 -DCIRCUITPY_BLE_FILE_SERVICE=0 -DCIRCUITPY_SERIAL_UART=0 -DCIRCUITPY_ULAB=1 -DCIRCUITPY_WATCHDOG=0 -DCIRCUITPY_ENABLE_MPY_NATIVE=0 -DINTERNAL_FLASH_FILESYSTEM=1 -DQSPI_FLASH_FILESYSTEM=0 -DSPI_FLASH_FILESYSTEM=0 -DUSB_AVAILABLE -DLONGINT_IMPL_MPZ -Os -DNDEBUG -DCFG_TUSB_MCU=OPT_MCU_SAMD51 -DCFG_TUD_MIDI_RX_BUFSIZE=128 -DCFG_TUD_CDC_RX_BUFSIZE=256 -DCFG_TUD_MIDI_TX_BUFSIZE=128 -DCFG_TUD_CDC_TX_BUFSIZE=256 -DCFG_TUD_MSC_BUFSIZE=1024 -flto -flto-partition=none -I. -I../.. -I../lib/mp-readline -I../lib/timeutils -Iasf4/samd51 -Iasf4/samd51/hal/include -Iasf4/samd51/hal/utils/include -Iasf4/samd51/hri -Iasf4/samd51/hpl/core -Iasf4/samd51/hpl/gclk -Iasf4/samd51/hpl/pm -Iasf4/samd51/hpl/port -Iasf4/samd51/hpl/rtc -Iasf4/samd51/hpl/tc -Iasf4/samd51/include -Iasf4/samd51/CMSIS/Include -Iasf4_conf/samd51 -Iboards/0_m4 -Iboards/ -Iperipherals/ -Ifreetouch -I../../lib/tinyusb/src -I../../supervisor/shared/usb -Ibuild-0_m4 -I../../drivers/wiznet5k -Wall -Werror -std=gnu11 -nostdlib -fshort-enums -fsingle-precision-constant -fno-strict-aliasing -Wdouble-promotion -Wno-endif-labels -Wstrict-prototypes -Werror-implicit-function-declaration -Wfloat-equal -Wundef -Wshadow -Wwrite-strings -Wsign-compare -Wmissing-format-attribute -Wno-deprecated-declarations -Wnested-externs -Wunreachable-code -Wcast-align -Wno-error=lto-type-mismatch -D__SAMD51G19A__ -ffunction-sections -fdata-sections -DCIRCUITPY_SOFTWARE_SAFE_MODE=0x0ADABEEF -DCIRCUITPY_CANARY_WORD=0xADAF00 -DCIRCUITPY_SAFE_RESTART_WORD=0xDEADBEEF --param max-inline-insns-single=500 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DCIRCUITPY_ULAB=1 -DMODULE_ULAB_ENABLED=1 -DMICROPY_PY_WIZNET5K=5500 -D_WIZCHIP_=5500  -mthumb -mabi=aapcs-linux -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DSAMD51 -DMICROPY_PY_NETWORK=1 -nostartfiles -Wl,-nostdlib -Wl,-T,build-0_m4/common.ld -Wl,-Map=build-0_m4/firmware.elf.map -Wl,-cref -Wl,-gc-sections -specs=nano.specs -mthumb -mcpu=cortex-m4 build-0_m4/py/mpstate.o build-0_m4/py/nlr.o build-0_m4/py/nlrx86.o build-0_m4/py/nlrx64.o build-0_m4/py/nlrthumb.o build-0_m4/py/nlrxtensa.o build-0_m4/py/nlrsetjmp.o build-0_m4/py/malloc.o build-0_m4/py/gc.o build-0_m4/py/gc_long_lived.o build-0_m4/py/pystack.o build-0_m4/py/qstr.o build-0_m4/py/vstr.o build-0_m4/py/mpprint.o build-0_m4/py/unicode.o build-0_m4/py/mpz.o build-0_m4/py/reader.o build-0_m4/py/lexer.o build-0_m4/py/parse.o build-0_m4/py/scope.o build-0_m4/py/compile.o build-0_m4/py/emitcommon.o build-0_m4/py/emitbc.o build-0_m4/py/asmbase.o build-0_m4/py/asmx64.o build-0_m4/py/emitnx64.o build-0_m4/py/asmx86.o build-0_m4/py/emitnx86.o build-0_m4/py/asmthumb.o build-0_m4/py/emitnthumb.o build-0_m4/py/emitinlinethumb.o build-0_m4/py/asmarm.o build-0_m4/py/emitnarm.o build-0_m4/py/asmxtensa.o build-0_m4/py/emitnxtensa.o build-0_m4/py/emitinlinextensa.o build-0_m4/py/formatfloat.o build-0_m4/py/parsenumbase.o build-0_m4/py/parsenum.o build-0_m4/py/emitglue.o build-0_m4/py/persistentcode.o build-0_m4/py/runtime.o build-0_m4/py/runtime_utils.o build-0_m4/py/scheduler.o build-0_m4/py/nativeglue.o build-0_m4/py/stackctrl.o build-0_m4/py/argcheck.o build-0_m4/py/warning.o build-0_m4/py/map.o build-0_m4/py/obj.o build-0_m4/py/objarray.o build-0_m4/py/objattrtuple.o build-0_m4/py/objbool.o build-0_m4/py/objboundmeth.o build-0_m4/py/objcell.o build-0_m4/py/objclosure.o build-0_m4/py/objcomplex.o build-0_m4/py/objdeque.o build-0_m4/py/objdict.o build-0_m4/py/objenumerate.o build-0_m4/py/objexcept.o build-0_m4/py/objfilter.o build-0_m4/py/objfloat.o build-0_m4/py/objfun.o build-0_m4/py/objgenerator.o build-0_m4/py/objgetitemiter.o build-0_m4/py/objint.o build-0_m4/py/objint_longlong.o build-0_m4/py/objint_mpz.o build-0_m4/py/objlist.o build-0_m4/py/objmap.o build-0_m4/py/objmodule.o build-0_m4/py/objobject.o build-0_m4/py/objpolyiter.o build-0_m4/py/objproperty.o build-0_m4/py/objnone.o build-0_m4/py/objnamedtuple.o build-0_m4/py/objrange.o build-0_m4/py/objreversed.o build-0_m4/py/objset.o build-0_m4/py/objsingleton.o build-0_m4/py/objslice.o build-0_m4/py/objstr.o build-0_m4/py/objstrunicode.o build-0_m4/py/objstringio.o build-0_m4/py/objtuple.o build-0_m4/py/objtype.o build-0_m4/py/objzip.o build-0_m4/py/opmethods.o build-0_m4/py/proto.o build-0_m4/py/reload.o build-0_m4/py/sequence.o build-0_m4/py/stream.o build-0_m4/py/binary.o build-0_m4/py/builtinimport.o build-0_m4/py/builtinevex.o build-0_m4/py/builtinhelp.o build-0_m4/py/modarray.o build-0_m4/py/modbuiltins.o build-0_m4/py/modcollections.o build-0_m4/py/modgc.o build-0_m4/py/modio.o build-0_m4/py/modmath.o build-0_m4/py/modcmath.o build-0_m4/py/modmicropython.o build-0_m4/py/modstruct.o build-0_m4/py/modsys.o build-0_m4/py/moduerrno.o build-0_m4/py/modthread.o build-0_m4/py/vm.o build-0_m4/py/bc.o build-0_m4/py/showbc.o build-0_m4/py/repl.o build-0_m4/py/smallint.o build-0_m4/py/frozenmod.o build-0_m4/py/ringbuf.o build-0_m4/extmod/moductypes.o build-0_m4/extmod/modujson.o build-0_m4/extmod/modure.o build-0_m4/extmod/moduzlib.o build-0_m4/extmod/moduheapq.o build-0_m4/extmod/modutimeq.o build-0_m4/extmod/moduhashlib.o build-0_m4/extmod/modubinascii.o build-0_m4/extmod/virtpin.o build-0_m4/extmod/modussl_axtls.o build-0_m4/extmod/modussl_mbedtls.o build-0_m4/extmod/modurandom.o build-0_m4/extmod/moduselect.o build-0_m4/extmod/modwebsocket.o build-0_m4/extmod/modwebrepl.o build-0_m4/extmod/modframebuf.o build-0_m4/extmod/vfs.o build-0_m4/extmod/vfs_reader.o build-0_m4/extmod/vfs_posix.o build-0_m4/extmod/vfs_posix_file.o build-0_m4/extmod/vfs_fat.o build-0_m4/extmod/vfs_fat_diskio.o build-0_m4/extmod/vfs_fat_file.o build-0_m4/extmod/utime_mphal.o build-0_m4/extmod/uos_dupterm.o build-0_m4/lib/embed/abort_.o build-0_m4/lib/utils/printf.o build-0_m4/main.o build-0_m4/supervisor/port.o build-0_m4/supervisor/shared/autoreload.o build-0_m4/supervisor/shared/board.o build-0_m4/supervisor/shared/filesystem.o build-0_m4/supervisor/shared/flash.o build-0_m4/supervisor/shared/micropython.o build-0_m4/supervisor/shared/rgb_led_status.o build-0_m4/supervisor/shared/safe_mode.o build-0_m4/supervisor/shared/stack.o build-0_m4/supervisor/shared/status_leds.o build-0_m4/supervisor/shared/tick.o build-0_m4/supervisor/shared/translate.o build-0_m4/supervisor/internal_flash.o build-0_m4/lib/tinyusb/src/common/tusb_fifo.o build-0_m4/lib/tinyusb/src/device/usbd.o build-0_m4/lib/tinyusb/src/device/usbd_control.o build-0_m4/lib/tinyusb/src/class/msc/msc_device.o build-0_m4/lib/tinyusb/src/class/cdc/cdc_device.o build-0_m4/lib/tinyusb/src/tusb.o build-0_m4/supervisor/shared/serial.o build-0_m4/supervisor/usb.o build-0_m4/supervisor/shared/usb/usb_desc.o build-0_m4/supervisor/shared/usb/usb.o build-0_m4/supervisor/shared/usb/usb_msc_flash.o build-0_m4/build-0_m4/autogen_usb_descriptor.o build-0_m4/lib/tinyusb/src/class/hid/hid_device.o build-0_m4/shared-bindings/usb_hid/__init__.o build-0_m4/shared-bindings/usb_hid/Device.o build-0_m4/shared-module/usb_hid/__init__.o build-0_m4/sh [... cut ...]
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld:build-0_m4/common.ld:22: syntax error
collect2: error: ld returned 1 exit status
Makefile:325: recipe for target 'build-0_m4/firmware.elf' failed
make: *** [build-0_m4/firmware.elf] Error 1
##[error]Bash exited with code '2'.
```

Here is the common.ld file generated : [common.ld.zip](https://github.com/adafruit/circuitpython/files/5898408/common.ld.zip)
I didn't saw anything obvious but I don't have any experience with linker script.

Any help will be apreciated :)
Thanks

ladyada, 2021-01-30:
can you submit a draft PR for the board - that was we see the config (and then we could also get it included anyways)

dhalbert, 2021-01-30:
I have a vague memory of trouble in the past when I used the division operator `/` in an `ld` script. Try rewriting any constants you specified without using division. But please also let's see your configuration files, as @ladyada suggests.

gc-elec, 2021-01-30:
Thanks for your quick reply, I just finished preparing the requested [PR draft](https://github.com/adafruit/circuitpython/pull/4100).

I'll try removing the divisions from the ld script, but I'm not sure where I should do this so that I don't break anything for the other boards?

dhalbert, 2021-01-30:
What does `arm-none-eabi-ld --version` print? I think this may have a been a bug in `ld` that was fixed.

gc-elec, 2021-01-30:
Here is the version text from ld :

`
GNU ld (2.27-9ubuntu1+9) 2.27
Copyright (C) 2016 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
`

dhalbert, 2021-01-30:
Yes :smile: . You should be using the ARM 2010 tool chain. Did you read and follow https://learn.adafruit.com/building-circuitpython ?

gc-elec, 2021-01-30:
Yes, but it was long ago when I build 4.x for SAMD21
I just updated the building environment to
`GNU ld (2.34-4ubuntu1+13ubuntu1) 2.34
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
`
No more syntax error 👍 
But now I get `region `FLASH_FIRMWARE' overflowed by 116560 bytes`
I can see that during the build that -DCIRCUITPY_FULL_BUILD=1 is passed to gcc, but I have CIRCUITPY_SMALL_BUILD = 1 in ma config.
Is the order important?

Should I close this issue and open a new one about the overflow?

dhalbert, 2021-01-30:
`CIRCUITPY_SMALL_BUILD` no long is functional. So you'll need to turn off modules you don't need in `mpconfigboard.h`. You can leave `CIRCUITPY_FULL_BUILD` on or turn it off and then turn on the modules you want. It's probably easier to leave it on and disable modules. Or you could reduce the size of the internal filesystem.

dhalbert, 2021-01-30:
> Should I close this issue and open a new one about the overflow?

This has turned into a support issue. We'll just close it when your build succeeds.

gc-elec, 2021-01-30:
I disabled a lot of modules (almost everything I found in circuitpy_mpconfig.h) but I still have an overflow and undefined references to neopixel :
`
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: build-0_m4/firmware.elf section `.text' will not fit in region `FLASH_FIRMWARE'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `FLASH_FIRMWARE' overflowed by 68472 bytes
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: firmware.elf.lto.o: in function `clear_temp_status':
<artificial>:(.text.clear_temp_status+0xc): undefined reference to `common_hal_neopixel_write'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: firmware.elf.lto.o: in function `new_status_color':
<artificial>:(.text.new_status_color+0x2e): undefined reference to `common_hal_neopixel_write'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: firmware.elf.lto.o: in function `temp_status_color':
<artificial>:(.text.temp_status_color+0x26): undefined reference to `common_hal_neopixel_write'
collect2: error: ld returned 1 exit status
make: *** [Makefile:326: build-0_m4/firmware.elf] Error 1
##[error]Bash exited with code '2'.
`

Did I missed something?

dhalbert, 2021-01-30:
You are still using gcc9. Upgrade to the gcc10 toolchain, as suggested. That will gain you some (not a lot) of space. Disable `CIRCUITPY_ULAB`. Take a look at the other disabled modules for similar boards to yours, such as `kicksat-sprite`.

Note this important change:
```
# Override optimization to keep binary small
OPTIMIZATION_FLAGS = -Os
```

SAMD51 is now normally compiled with -O2, which makes for considerably larger code.

You disabled `neopixel_write`, which is needed for the status NeoPixel. Don't disable that.

gc-elec, 2021-01-31:
I followed the guide again to use the exact version of the toolchain, and added the optimization flags but I still exceed the allowed size by almost 13 kb
```
gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region 'FLASH_FIRMWARE' overflowed by 12984 bytes
```
I disabled more modules than kicksat-sprite but still oversize :(

I also noticed a lot of multiple definitions notices from ld, is this expected?
```
gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: build-0_m4/drivers/wiznet5k/ethernet/socket.o (symbol from plugin): in function `wizchip_socket_reset':
(.text+0x0): multiple definition of `audio_dma_in_use'; build-0_m4/py/mpstate.o (symbol from plugin):(.text+0x0): first defined here
```

gc-elec, 2021-01-31:
I followed the guide again and then noticed I did a mistake when rebasing my local repo.
Now everything is clean and up to date, but I'm still oversized by 39 kb :(

```
arm-none-eabi-gcc -o build-0_m4/firmware.elf -DCIRCUITPY_FULL_BUILD=1 -DMICROPY_PY_ASYNC_AWAIT=1 -DCIRCUITPY_AESIO=0 -DCIRCUITPY_ALARM=0 -DCIRCUITPY_ANALOGIO=1 -DCIRCUITPY_AUDIOBUSIO=0 -DCIRCUITPY_AUDIOBUSIO_I2SOUT=0 -DCIRCUITPY_AUDIOBUSIO_PDMIN=0 -DCIRCUITPY_AUDIOIO=1 -DCIRCUITPY_AUDIOPWMIO=0 -DCIRCUITPY_AUDIOCORE=1 -DCIRCUITPY_AUDIOMIXER=1 -DCIRCUITPY_AUDIOMP3=1 -DCIRCUITPY_BINASCII=1 -DCIRCUITPY_BITBANGIO=0 -DCIRCUITPY_BLEIO_HCI=1 -DCIRCUITPY_BLEIO=1 -DCIRCUITPY_BOARD=1 -DCIRCUITPY_BUSDEVICE=1 -DCIRCUITPY_BUILTINS_POW3=1 -DCIRCUITPY_BUSIO=1 -DCIRCUITPY_CAMERA=0 -DCIRCUITPY_CANIO=0 -DCIRCUITPY_DIGITALIO=1 -DCIRCUITPY_COUNTIO=1 -DCIRCUITPY_DISPLAYIO=0 -DCIRCUITPY_ERRNO=1 -DCIRCUITPY_ESPIDF=0 -DCIRCUITPY_FRAMEBUFFERIO=0 -DCIRCUITPY_VECTORIO=0 -DCIRCUITPY_FREQUENCYIO=0 -DCIRCUITPY_GAMEPAD=0 -DCIRCUITPY_GAMEPADSHIFT=0 -DCIRCUITPY_GNSS=0 -DCIRCUITPY_I2CPERIPHERAL=1 -DCIRCUITPY_IPADDRESS=0 -DCIRCUITPY_JSON=1 -DCIRCUITPY_MATH=1 -DCIRCUITPY__EVE=0 -DCIRCUITPY_MEMORYMONITOR=0 -DCIRCUITPY_MICROCONTROLLER=1 -DCIRCUITPY_NEOPIXEL_WRITE=1 -DCIRCUITPY_NETWORK=0 -DCIRCUITPY_NVM=1 -DCIRCUITPY_OS=1 -DCIRCUITPY_DUALBANK=0 -DCIRCUITPY_PIXELBUF=1 -DCIRCUITPY_PS2IO=0 -DCIRCUITPY_PULSEIO=0 -DCIRCUITPY_PWMIO=0 -DCIRCUITPY_RANDOM=1 -DCIRCUITPY_RE=1 -DCIRCUITPY_RP2PIO=0 -DCIRCUITPY_RGBMATRIX=0 -DCIRCUITPY_ROTARYIO=0 -DCIRCUITPY_RTC=1 -DCIRCUITPY_SAMD=1 -DCIRCUITPY_SDCARDIO=1 -DCIRCUITPY_SDIOIO=0 -DCIRCUITPY_SHARPDISPLAY=0 -DCIRCUITPY_SOCKETPOOL=0 -DCIRCUITPY_SSL=0 -DCIRCUITPY_STAGE=0 -DCIRCUITPY_STORAGE=1 -DCIRCUITPY_STRUCT=1 -DCIRCUITPY_SUPERVISOR=1 -DCIRCUITPY_TERMINALIO=0 -DCIRCUITPY_TIME=1 -DCIRCUITPY_TOUCHIO_USE_NATIVE=0 -DCIRCUITPY_TOUCHIO=1 -DCIRCUITPY_UHEAP=0 -DCIRCUITPY_USB_HID=1 -DCIRCUITPY_USB_MIDI=1 -DCIRCUITPY_PEW=0 -DCIRCUITPY_USTACK=0 -DCIRCUITPY_BITBANG_APA102=0 -DCIRCUITPY_REQUIRE_I2C_PULLUPS=1 -DCIRCUITPY_SERIAL_BLE=0 -DCIRCUITPY_BLE_FILE_SERVICE=0 -DCIRCUITPY_SERIAL_UART=0 -DCIRCUITPY_ULAB=0 -DCIRCUITPY_WATCHDOG=0 -DCIRCUITPY_WIFI=0 -DCIRCUITPY_ENABLE_MPY_NATIVE=0 -DCIRCUITPY_MSGPACK=1 -DINTERNAL_FLASH_FILESYSTEM=1 -DQSPI_FLASH_FILESYSTEM=0 -DSPI_FLASH_FILESYSTEM=0 -DUSB_AVAILABLE -DCIRCUITPY_DEBUG=0 -DLONGINT_IMPL_MPZ -DCFG_TUSB_MCU=OPT_MCU_SAMD51 -DCFG_TUD_MIDI_RX_BUFSIZE=128 -DCFG_TUD_CDC_RX_BUFSIZE=256 -DCFG_TUD_MIDI_TX_BUFSIZE=128 -DCFG_TUD_CDC_TX_BUFSIZE=256 -DCFG_TUD_MSC_BUFSIZE=1024 -Os -ftree-vrp -DNDEBUG -flto -flto-partition=none -I../../devices/ble_hci -I. -I../.. -I../lib/mp-readline -I../lib/timeutils -Iasf4/samd51 -Iasf4/samd51/hal/include -Iasf4/samd51/hal/utils/include -Iasf4/samd51/hri -Iasf4/samd51/hpl/core -Iasf4/samd51/hpl/gclk -Iasf4/samd51/hpl/pm -Iasf4/samd51/hpl/port -Iasf4/samd51/hpl/rtc -Iasf4/samd51/hpl/tc -Iasf4/samd51/include -Iasf4/samd51/CMSIS/Include -Iasf4_conf/samd51 -Iboards/0_m4 -Iboards/ -Iperipherals/ -Ifreetouch -I../../lib/tinyusb/src -I../../supervisor/shared/usb -Ibuild-0_m4 -Wall -Werror -std=gnu11 -nostdlib -fshort-enums -fsingle-precision-constant -fno-strict-aliasing -Wdouble-promotion -Wimplicit-fallthrough=2 -Wno-endif-labels -Wstrict-prototypes -Werror-implicit-function-declaration -Wfloat-equal -Wundef -Wshadow -Wwrite-strings -Wsign-compare -Wmissing-format-attribute -Wno-deprecated-declarations -Wnested-externs -Wunreachable-code -Wcast-align -D__SAMD51G19A__ -ffunction-sections -fdata-sections -DCIRCUITPY_SOFTWARE_SAFE_MODE=0x0ADABEEF -DCIRCUITPY_CANARY_WORD=0xADAF00 -DCIRCUITPY_SAFE_RESTART_WORD=0xDEADBEEF -DCIRCUITPY_BOARD_ID="\"0_m4\"" --param max-inline-insns-single=500 -DFFCONF_H=\"lib/oofatfs/ffconf.h\"  -mthumb -mabi=aapcs-linux -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DSAM_D5X_E5X -DSAMD51 -nostartfiles -Wl,-nostdlib -Wl,-T,build-0_m4/common.ld -Wl,-Map=build-0_m4/firmware.elf.map -Wl,-cref -Wl,-gc-sections -specs=nano.specs -flto=2 -mthumb -mcpu=cortex-m4 build-0_m4/py/mpstate.o build-0_m4/py/nlr.o build-0_m4/py/nlrx86.o build-0_m4/py/nlrx64.o build-0_m4/py/nlrthumb.o build-0_m4/py/nlrxtensa.o build-0_m4/py/nlrsetjmp.o build-0_m4/py/malloc.o build-0_m4/py/gc.o build-0_m4/py/gc_long_lived.o build-0_m4/py/pystack.o build-0_m4/py/qstr.o build-0_m4/py/vstr.o build-0_m4/py/mpprint.o build-0_m4/py/unicode.o build-0_m4/py/mpz.o build-0_m4/py/reader.o build-0_m4/py/lexer.o build-0_m4/py/parse.o build-0_m4/py/scope.o build-0_m4/py/compile.o build-0_m4/py/emitcommon.o build-0_m4/py/emitbc.o build-0_m4/py/asmbase.o build-0_m4/py/asmx64.o build-0_m4/py/emitnx64.o build-0_m4/py/asmx86.o build-0_m4/py/emitnx86.o build-0_m4/py/asmthumb.o build-0_m4/py/emitnthumb.o build-0_m4/py/emitinlinethumb.o build-0_m4/py/asmarm.o build-0_m4/py/emitnarm.o build-0_m4/py/asmxtensa.o build-0_m4/py/emitnxtensa.o build-0_m4/py/emitinlinextensa.o build-0_m4/py/formatfloat.o build-0_m4/py/parsenumbase.o build-0_m4/py/parsenum.o build-0_m4/py/emitglue.o build-0_m4/py/persistentcode.o build-0_m4/py/runtime.o build-0_m4/py/runtime_utils.o build-0_m4/py/scheduler.o build-0_m4/py/nativeglue.o build-0_m4/py/stackctrl.o build-0_m4/py/argcheck.o build-0_m4/py/warning.o build-0_m4/py/map.o build-0_m4/py/enum.o build-0_m4/py/obj.o build-0_m4/py/objarray.o build-0_m4/py/objattrtuple.o build-0_m4/py/objbool.o build-0_m4/py/objboundmeth.o build-0_m4/py/objcell.o build-0_m4/py/objclosure.o build-0_m4/py/objcomplex.o build-0_m4/py/objdeque.o build-0_m4/py/objdict.o build-0_m4/py/objenumerate.o build-0_m4/py/obje [... cut ...] roc.o build-0_m4/lib/mp3/src/subband.o build-0_m4/lib/mp3/src/trigtabs.o -Wl,--start-group -lgcc -lc -Wl,--end-group
/home/vsts/work/1/s/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: build-0_m4/firmware.elf section `.text' will not fit in region `FLASH_FIRMWARE'
/home/vsts/work/1/s/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH_FIRMWARE' overflowed by 39644 bytes
```

dhalbert, 2021-01-31:
Push to your own repo and I will take a quick look. You only have 512kB of flash, so only 256kB for firmware, so you want to choose modules that, say, an M0 Express would pick.

gc-elec, 2021-01-31:
Ok, I pushed to my repo with the latests changes.
I saw that the PR triggered a new build on your side which shows the failing and the 39 kb oversize.
I disabled the I saw disabled for the kicksat-sprite but I missed CIRCUITPY_AUDIOMP3 and CIRCUITPY_BLEIO_HCI.
I'll try desabling these two but I could need CIRCUITPY_AUDIOMP3 later.

gc-elec, 2021-01-31:
Aaaaand it PASS :)
With CIRCUITPY_AUDIOMP3 and CIRCUITPY_BLEIO_HCI disabled, the build passed.
I pushed to my repo and I'll try it on the board.

Thank you very much for your time and help :)

dhalbert, 2021-01-31:
Great! Before you make the final PR, could you add a "manufacturer", and change the board name to something more descriptive? Also maybe do some rebasing to clean up the commits (and/or start a new PR).

gc-elec, 2021-01-31:
Yes, I will clean up everything.
I didn't planned to do the final PR yet because the board is still in an early stage (need to fix some mistakes on the PCB and pin assignments) and I will have to troubleshoot the flash chips I have (only one out of 3 boards have the flash chip detected, I don't know why yet).

gc-elec, 2021-02-01:
Hello again
Now that the internal flash build works and the board is usable, I'm trying to make the external SPI flash operational.

I'm not sure if I could ask here or open a new issue?

I use the GD25Q64CSIG chip connected with the same pins as on the Feather M0 Express :
[image]

I successfully used this setup with a SAMD21, but it looks like the SAMD51 does not detect the chip (the CIRCUITPY drive does not show up when I enable external flash).

I build 2 of these board with SAMD21, and one with SAMD51.
Only one with SAMD21 have a working flash.
Could it be that I killed the flash chips using a ohmmeter while checking the solderings?

gc-elec, 2021-02-05:
So, I hooked up a small test bench using a raspberry pi to test one of the flash chip and found out it is OK.
I tried to attach it to the SAMD51 board without success.

I searched a bit more about SPI on circuitPyhton and found out [this](https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/creating-a-new-spi).
According to the datasheet, the wiring I used works for SAMD21, but not for SAMD51 due to the restriction on SERCOM pads.

I checked is using your handy script [here](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/CircuitPython_Essentials/SPI_Test_Script.py) and it's the problem.

So, the solution was swapping MOSI and SCK (physically on the PCB by hacking a bit and in mpconfigboard.h) and tadaaa M4 express variant!
[image]

I will swap MOSI and SCK on the PCB design for an hypothetical V2 so it can be built with a SAMD21 or a SAMD51 indifferently :)

closed 2021-03-24 by tannewt