adafruit/circuitpython #4101: SocketPool.Socket.__hash__() is not immutable

dhalbert, 2021-01-30:
`__hash__()` was added to `Socket` so a `Socket` could be put in a dict, for use in the `adafruit_requests` library. However, the hash used is not immutable, and changes when the socket is closed.

The hash returned could be `id()`, to make it immutable.. However, it might be worth looking at whether the use of `Socket` in `adafruit_requests` really needs a dictionary. I see two uses: `._open_sockets` and `._socket_free`. I'll bring this up in an issue in that repo.

Tagging @hierophect on this, who was trying to debug #4061, which may be caused by this.

hierophect, 2021-01-30:
@dhalbert as a follow up to our conversation, just noting that returning `(mp_uint_t)self` for `common_hal_ssl_sslsocket_get_hash` does solve the keys issue in #4061, as predicted (I don't know if that's how I'm supposed to code it though, can I use `id` in the shared-bindings layer?). You mentioned you were concerned about other instances where the ID could change unexpectedly, making the hash mutable, were you able to find those issues?

dhalbert, 2021-01-30:
The only other issue is this: #2687, which I think is not applicable. You could try the fix above, and see how it works.

You can can return `mp_obj_id()` to get the id value for the hash. I would think it's OK to do that in shared-bindings, because calling that function is kind of like calling a common-hal function: it's executing the value.

dhalbert, 2021-01-30:
In CPython sockets, `socket.fileno()` will return -1 if the socket is closed. So we could implement that (just return `.num`).

hierophect, 2021-02-01:
@dhalbert you mean, keep the hash as returning `id()` so it isn't mutable, but implement `socket.fileno()` so we can still access the socket number? I think `fileno` was on my list anyway, along some of the other "filler" functions in socket.

dhalbert, 2021-02-01:
Yes, keep using `id()`. But if we implement `.fileno()`, can we take advantage of the fact that it returns `-1` when the socket is closed, rather than keeping local state about that? It's really a question about what the requests library can do with it.

hierophect, 2021-02-01:
I tried using `hash()` on Cpython's sockets, and it seems to work fine. It's not related to the result of `fileno` of course, as that's also mutable in Cpython. I can't find any takes online about whether it's advisable to use Sockets as keys, though.

hierophect, 2021-02-01:
@dhalbert if you mean internal state within the Requests library, I think so, yes. We create a `sock->num` in C when the socket is created, and set it to -1 when closed, and that's all that `fileno` would be returning. The only potential issue is that TLS sockets don't strictly need to have the original socket open to keep working - in Scott's original implementation, they didn't call the `lwip_socket` function at all, which is what allocates the file descriptor. But my implementation calls the LWIP initializer and doesn't close it, because it's the only way to give SSLSocket access to bind/listen/accept (it's based off the CPython implementation that does the same thing). So `fileno() = -1` should basically always be an indicator that the socket has been closed.

tannewt, 2021-02-02:
@hierophect Can you open the second LWIP socket only when you do the server APIs?

PR #4049 (2021-02-02): Separate SSLSocket from Socket
This PR removes the TLS implementations from Socket and moves them to a new object class, SSLSocket, which is created from a Socket object using SSL's `contex.wrap_socket()` function. This behavior more closely matches that of [Cpython](https://docs.python.org/3/library/ssl.html#ssl-sockets).

Opening as a draft as I've had a couple difficulties with this change and I'm still testing:

- The regular Socket class has a new LWIP-only implementation for `connect`, but I've had issues making this non-blocking. A non-blocking call to `connect` should always return a - [... cut]
closed 2021-02-02 by tannewt