adafruit/circuitpython #4103: ESP32-S2: Configurable resources based on board RAM

anecdata, 2021-01-30:
Is it possible to have multiple configs for RAM, like we do for flash `circuitpython/ports/esp32s2/esp-idf-config/` so that boards with more RAM could be allocated more resources, like more sockets, more pystack, etc.?

dhalbert, 2021-01-30:
The nRF ports do RAM allocation to set aside RAM for use by the nRF SoftDevice and to allocate RAM for a SPIM3 buffer (a hardware bug workaround):
https://github.com/adafruit/circuitpython/blob/70a5f464df709737fb90a9ab69baf08671c8f383/ports/nrf/boards/common.template.ld#L21-L31.

anecdata, 2021-01-30:
We have `CONFIG_SPIRAM_SIZE`  in board-by-board `sdkconfig`, but it seems if other resource parameters are board-by-board, capabilities may fragment and be harder to troubleshoot when issues arise.

tannewt, 2021-02-01:
Note that right now we don't use external ram for sockets or pystack. We simply use all external RAM for the CP heap. To allow for more sockets, we'd need to change how RAM is shared between CP and the IDF. Basically now we don't tell the IDF about it. In the future we could tell the IDF and then allocate the heap through the IDF.

One wrinkle is that then the IDF memory may not be DMA-able so for any IDF buffers we'll have to be careful.

hierophect, 2021-02-01:
This recently came up in #4061, since TLS sockets take up 2 sockets apiece now, so you can only have 2 of them total. @anecdata noted that might be a bit restrictive.

anecdata, 2021-02-02:
Looks like `PYSTACK` is set to 1536 globally in `py/circuitpy_mpconfig.h`, but could be easily overridden at the port level (no existing overrides I could find) or board level (I think only TG-Watch does this).

In cases where there is sufficient external RAM for heap, what's the downside to using more internal RAM for IDF/sockets and pystack?

tannewt, 2021-02-04:
I think you'll find that the tradeoff is between IDF/sockets (which gets the remainder) and pystack.

anecdata, 2022-08-23:
More up-to-date detail, with references to PRs addressing some of the use cases, in #5915.

Closing this old issue.

closed 2022-08-23 by anecdata