adafruit/circuitpython #4107: Using GP15 on RPi Pico can prevent code.py from running automatically

dhalbert, 2021-02-01:
A user reworked [this example](https://learn.adafruit.com/circuitpython-essentials/circuitpython-hid-keyboard-and-mouse#circuitpython-keyboard-emulator-2985260-2) to use GP14 and GP15, and their `code.py` would not start automatically. Changing GP15 to another pin fixed the problem.

https://discord.com/channels/327254708534116352/537365702651150357/805826302699110421 and following.

tannewt, 2021-02-02:
I suspect that code.py was working but the GP15 button didn't work. The errata fix doesn't restore the mux setting last I checked. @kilograham, is this on your radar?

waiweng83, 2021-02-02:
I tried it out and confirmed that program using GP15 does not run when saved as code.py or main.py.
It also has some bad impact on the REPL.

However, saved program run from REPL seems to just work fine.

waiweng83, 2021-02-02:
> I suspect that code.py was working but the GP15 button didn't work. The errata fix doesn't restore the mux setting last I checked. @kilograham, is this on your radar?

@tannewt I tried building my own fork and playing around with the errata fix and seems like the call to this function is causing the problem.
```
// Force LS_J
const uint dp = 15;
//const uint dm = 16;

gpio_set_function(dp, 8);
```

Even I tried saving the original value prior to calling the function and restore it after, but it seems the problem is still there.

kilograham, 2021-02-02:
> I suspect that code.py was working but the GP15 button didn't work. The errata fix doesn't restore the mux setting last I checked. @kilograham, is this on your radar?

i had heard some mention from someone, not sure the specifics... can you open an SDK issue

tannewt, 2021-02-02:
Filed https://github.com/raspberrypi/pico-sdk/issues/76 . Thanks!

dhalbert, 2021-02-05:
pico-sdk PR https://github.com/raspberrypi/pico-sdk/pull/80 is in progress and addresses this

lurch, 2021-02-05:
Now fixed in the `develop` branch of pico-sdk, if you fancy testing it :slightly_smiling_face:

tannewt, 2021-02-05:
@lurch Thanks! I'll update us to that commit and get it into a beta for testing.

dhalbert, 2021-02-07:
@waiweng83 If you have time, could you check https://adafruit-circuit-python.s3.amazonaws.com/bin/raspberry_pi_pico/en_US/adafruit-circuitpython-raspberry_pi_pico-en_US-20210206-a10ce39.uf2 or later for this problem. It is from #4141 and incorporates a fix in the pico-sdk. Thanks.

dhalbert, 2021-02-07:
0x414b on discord reports the old program seems to be gone: https://discord.com/channels/327254708534116352/537365702651150357/808000045147750401

waiweng83, 2021-02-08:
> @waiweng83 If you have time, could you check https://adafruit-circuit-python.s3.amazonaws.com/bin/raspberry_pi_pico/en_US/adafruit-circuitpython-raspberry_pi_pico-en_US-20210206-a10ce39.uf2 or later for this problem? It is from #4141 and incorporates a fix in the pico-sdk. Thanks.

Yes, I tested with code.py and main.py and it's working perfectly now.

> 0x414b on discord reports the old program seems to be gone: https://discord.com/channels/327254708534116352/537365702651150357/808000045147750401

My old code seems like is still there. I don't have any problem with this.

dhalbert, 2021-02-08:
Destinys Agent on discord [also reports success](https://discord.com/channels/327254708534116352/537365702651150357/808264323372679199). We can close this as fixed by #4141.

closed 2021-02-08 by dhalbert