adafruit/circuitpython #4108: ESP32-S2: Getting this error while running the SD card demo 'SPIDevice' object has no attribute 'spi'

eddieespinal, 2021-02-01:
Hi there,

I'm currently testing a new board that uses the ESP32-S2 and has an integrated SD card. When I try using one of the existing board from the website it works fine but when I flash it with my own board definition the SD card won't work and I get the following error message:

```
Traceback (most recent call last):
  File "code.py", line 16, in <module>
  File "adafruit_sdcard.py", line 101, in __init__
  File "adafruit_sdcard.py", line 122, in _init_card
  File "adafruit_sdcard.py", line 110, in _clock_card
AttributeError: 'SPIDevice' object has no attribute 'spi'
```

This is the code I'm trying to run:

```
import time

import adafruit_sdcard
import board
import busio
import digitalio
import microcontroller
import storage

# Use any pin that is not taken by SPI
SD_CS = board.SD_CS

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# Use the filesystem as normal! Our files are under /sd

print("Logging temperature to filesystem")
# append to the file!
while True:
    # open file for append
    with open("/sd/temperature.txt", "a") as f:
        t = microcontroller.cpu.temperature
        print("Temperature = %0.1f" % t)
        f.write("%0.1f\n" % t)
    # file is saved
    time.sleep(1)

```

The output of my terminal after running: `make V=1 BOARD=atmegazero_esp32s2 flash PORT=/dev/tty.SLAB_USBtoUART ` is:

```
[1/1] Performing build step for 'bootloader'
ninja: no work to do.
LINK build-atmegazero_esp32s2/firmware.elf
# @xtensa-esp32s2-elf-size build-atmegazero_esp32s2/firmware.elf | python3 ../../tools/build_memory_info.py build-atmegazero_esp32s2/esp-idf/esp-idf/esp32s2/esp32s2_out.ld
Create build-atmegazero_esp32s2/circuitpython-firmware.bin
esptool.py v3.0
esptool.py --chip esp32s2 -p /dev/tty.SLAB_USBtoUART -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode qio --flash_freq 40m --flash_size 16MB 0x0000 build-atmegazero_esp32s2/firmware.bin
esptool.py v3.0
Serial port /dev/tty.SLAB_USBtoUART
Connecting........___
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: 7c:df:a1:00:3e:af
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 1359088 bytes to 860755...
Wrote 1359088 bytes (860755 compressed) at 0x00000000 in 21.4 seconds (effective 508.1 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```

What am I missing?

Thanks for your help

tannewt, 2021-02-01:
I think this is a bug on our end. I believe beta.0 should work in the meantime.

eddieespinal, 2021-02-01:
Thank you for the response @tannewt, that explains why this build worked `adafruit-circuitpython-unexpectedmaker_feathers2-en_US-6.2.0-beta.0.bin` but not if I compile it using make for the same board. 

I will give it a try with beta.0 and see if that helps.

Thanks again!

eddieespinal, 2021-02-01:
@tannewt you were right, that was the problem. After trying again using the branch `6.1.x` everything works fine now.

Thanks for solving my problem.

jfabernathy, 2021-02-01:
I get this same error on both ESP32-S2 and Raspberry PI pico, but works on Seeeduio Xiao with same version of CP

tannewt, 2021-02-02:
> I get this same error on both ESP32-S2 and Raspberry PI pico, but works on Seeeduio Xiao with same version of CP

I think the Xiao works because it uses the python version of bus device. The other two have it built in.

closed 2021-02-02 by tannewt

polygnomial, 2021-05-29:
Hello, I'm seeing this error on PyCubed board in both 6.2 and 7.0 Alpha 1 versions of CircuitPython, any ideas? Thanks!

tannewt, 2021-06-01:
@polygnomial Make sure you have the latest version of the SDCard library. IIRC we stopped using the .spi property from the SPIDevice class.

PR #4114 (2021-02-02): Add .spi accessor to SPIDevice
Fixes #4108
commit 8fd6bff7 (2021-02-02): Add .spi accessor to SPIDevice

Fixes #4108