adafruit/circuitpython #4111: pulseio module missing

eternaltinkering, 2021-02-01:
Adafruit CircuitPython 6.2.0-beta.1 on 2021-01-27; Raspberry Pi Pico with rp2040

>>> import simpleio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "simpleio.py", line 34, in <module>
ImportError: no module named 'pulseio'

dhalbert, 2021-02-01:
There is no `pulseio` yet for CircuitPython on the RP2040. `simpleio` imports `pulseio`, but usually you just need `simpleio` for the `map_range()` function it defines. You can copy that function and add it to your program for now.

Is this an Adafruit-supplied example? We can change the example.

eternaltinkering, 2021-02-01:
Oddly enough the simpleio module is used as an example for import errors in the getting started tutorial, mostly wanted to make sure I understood why I was getting the error after adding simpleio.mpy to the lib folder on the device.
https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-libraries#example-importerror-due-to-missing-library-2978247-20

kevinjwalters, 2021-02-15:
BTW, `PWMOut` is documented as being in pulseio until 7.x: https://circuitpython.readthedocs.io/en/6.1.x/shared-bindings/pulseio/#pulseio.PWMOut

tannewt, 2021-02-16:
PWMOut was not made available in pulseio because the rest of pulseio hasn't been implemented on the RP2040. All new code should use it from `pwmio` anyway (`pulseio.PWMOut` is deprecated.)

jwhevans, 2021-02-21:
Where is pulseio located? Is this coming from MicroPython? I haven't been able to find it.

dhalbert, 2021-02-21:
See `pulseio` and `pwmio` in this list: https://circuitpython.readthedocs.io/en/latest/shared-bindings/index.html

jwhevans, 2021-02-21:
Thank you. That was helpful. Somehow I overlooked it. I am lost as to how these modules build into the final CircuitPython file for the board. I see the implementation in the Shared Bindings for PulseIO but can't follow how these shared bindings are connected (well how they would be connected in the case of the RP2040) to the rest of the code. Can someone offer guidance here?

dhalbert, 2021-02-21:
@jwhevans This may be slightly out of date but the basic concepts are still true: https://circuitpython.readthedocs.io/en/latest/docs/common_hal.html

tannewt, 2021-02-22:
> Can someone offer guidance here?

The mechanics for adding a module to the build and putting it in the top level modules is in `py/circuitpy-`. It's controlled by the `CIRCUITPY_PULSEIO` definition. Once it's enabled, the build will fail because `ports/raspberrypi/common-hal/pulseio` is missing. That's where the implementation goes.

DavePutz, 2021-03-17:
pulsein for the RP2040 was addressed by PR #4262. pulseout is still not implemented.

DavePutz, 2021-06-09:
This was addressed by PR #4773 - Implement pulseout for RP2040.

tannewt, 2021-06-09:
Thanks @DavePutz!

closed 2021-06-09 by tannewt