adafruit/circuitpython #4115: ESP32-S2: hangs during startup if backlight pin is used in display init code

m4tk, 2021-02-02:
Hi there,

I'm currently trying to write the display init code for the newly merged ESP32-S2 LilyGo T8 board with ST7789 display.

In general it's working so far. But I'm struggeling somehow with the backlight pin. If I've understood the whole thing correctly it should be enough to pass the adress of the mcu_pin_object as parameter to the `common_hal_displayio_display_construct()` function.

But if I do that the board is hanging at startup. If I pass `NULL` as backlight pin and set the PIN "manually" to high as shown in the workaround code below it's all working fine.

```C
common_hal_displayio_display_construct(
      display,
      bus
      ...
      NULL,  // backlight pin ---->  &pin_GPIO33
      ...
);

/* workaround for initializing backlight */
digitalio_digitalinout_obj_t backlight_inout;
common_hal_digitalio_digitalinout_construct(&backlight_inout, &pin_GPIO33);
common_hal_digitalio_digitalinout_switch_to_output(&backlight_inout, true, DRIVE_MODE_PUSH_PULL);
common_hal_never_reset_pin(&pin_GPIO33);
```
Does anyone have any idea how to solve this issue?

many thanks in advance,

m4tk

[image]

tannewt, 2021-02-02:
I suspect this is a bug in the ESP32-S2 PWM implementation used by display. I'd be ok merging it with the digitalinout code for now and leaving this issue open for someone to look into it.

m4tk, 2021-02-02:
@tannewt Thanks Scott for your fast reply.

In the meanwhile I have looked a little deeper. It seems that `common_hal_pwmio_pwmout_construct()` function couldn't find a non-reserved channel because the `pwmout_reset()` function is called to late.

`pwmout_reset()` is called by `reset_port()`. But in main.c the `board_init()` which does the display initialization is called before `reset_port()`. So some initializations are not done yet.

I changed the order quick&dirty for testing in main.c...at least the board is not hanging during startup anymore. But something is still wrong there because the backlight is now short flashing (the text output is visible for a fraction of a second) but then leaves dark.

m4tk, 2021-02-03:
@tannewt Is it ok to call `pwmout_reset()` in the `board_init()`function to temporary solve the problem? if yes then I can try to create a PR (will be the very first 😯)

tannewt, 2021-02-04:
> @tannewt Is it ok to call `pwmout_reset()` in the `board_init()`function to temporary solve the problem? if yes then I can try to create a PR (will be the very first 😯)

Yup! (Usually the in use checks default to free.) If you find it gets reset you may need a `never_reset` call.

Thanks for debugging this!

m4tk, 2021-02-09:
solved for now by call of pwmout_reset() in the board_init() function.

closed 2021-02-09 by m4tk