adafruit/circuitpython #4117: Update pico-sdk to 1.0.1

tannewt, 2021-02-02:
Fixes a number of things and exposes a way to get a flash's unique id: https://github.com/raspberrypi/pico-sdk/releases/tag/1.0.1

ajs256, 2021-02-02:
I would like to work on this, could someone point me to what I would need to change? I see `/ports/raspberrypi/sdk_config/pico/`, is there anything else?

tannewt, 2021-02-02:
Epic! you'll need to change the commit referenced by ports/raspberrypi/sdk and then rebuild. Things may have changed so you'll likely need to update Makefile too. I think the linker script also changed so compare the CP copy to the sdk's default one.

ajs256, 2021-02-02:
Sorry, I'm having git troubles. I can't work on this anymore, please disregard my previous comment.

tannewt, 2021-02-02:
Thanks for trying @ajs256! I'll do it so I can get the correct timers for DMA.

dhalbert, 2021-02-08:
Superceded by #4141, which updates the `pico-sdk` to several commits past `1.0.1`.

closed 2021-02-08 by dhalbert