adafruit/circuitpython #4119: sdcard demo missing baudrate attribute on Raspberry Pi Pico

jfabernathy, 2021-02-03:
After #4114 was closed, I tested a Pico with adafruit-circuitpython-raspberry_pi_pico-en_US-20210202-9140bfb.uf2 and the latest library bundle as 20210203. I modified the pin number in the demo and it still draws and error:
`Traceback (most recent call last):
  File "code.py", line 23, in <module>
  File "adafruit_sdcard.py", line 101, in __init__
  File "adafruit_sdcard.py", line 122, in _init_card
  File "adafruit_sdcard.py", line 111, in _clock_card
AttributeError: 'SPIDevice' object has no attribute 'baudrate'`
My modified demo code is attached:
[code.txt](https://github.com/adafruit/circuitpython/files/5918448/code.txt)

jfabernathy, 2021-02-03:
I also get this error on ESP32-S2 Metro

tannewt, 2021-02-03:
@jfabernathy Can you set it to this and try it? (In `adafruit_sdcard.py`)

```python
self._spi.spi.configure(baudrate=250000)
```

Thanks for testing this!

jfabernathy, 2021-02-03:
I'm going to need a little help the adafruit_sdcard I have is the one in the bundle which is a mpy version. Where to I get the .py version?

jerryneedell, 2021-02-03:
Use the second button here : https://circuitpython.org/libraries

jerryneedell, 2021-02-03:
Or go to the actual repository  https://github.com/adafruit/Adafruit_CircuitPython_SD

jfabernathy, 2021-02-03:
So I got adafruit_sdcard.py and just put that in the CIRCUITPY/lib directory on the ESP32S2 and changed to `self._spi.spi.configure(baudrate=250000)`
Now I'm getting another error.
 ```
 File "code.py", line 18, in <module>
  File "/lib/adafruit_sdcard.py", line 101, in __init__
  File "/lib/adafruit_sdcard.py", line 122, in _init_card
  File "/lib/adafruit_sdcard.py", line 112, in _clock_card
AttributeError: 'SPIDevice' object has no attribute 'chip_select'
```

tannewt, 2021-02-03:
@jfabernathy try commenting line 112 out. The pin should already be high

jfabernathy, 2021-02-03:
with #112 commented out in adafruit_sdcard.py 
line 18 in demo program, `sdcard = adafruit_sdcard.SDCard(spi, cs)`
still give this error
```
  File "code.py", line 18, in <module>
  File "/lib/adafruit_sdcard.py", line 101, in __init__
  File "/lib/adafruit_sdcard.py", line 166, in _init_card
AttributeError: 'SPIDevice' object has no attribute 'chip_select'
```

tannewt, 2021-02-04:
Ok, looks like sdcard takes in the chip select so it should just remember it.