adafruit/circuitpython #4120: RP2040: charlcd demo fails

jerryneedell, 2021-02-03:
As reported by a Discord User, the accessing a Character LCD fails on a Raspberry Pi Pico

The same example runs normally on a metro-esp32s2

For this example I used SCL on GP1 SDA on GP0
 

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.1 on 2021-01-28; Raspberry Pi Pico with rp2040
>>> 
>>> import charlcd_i2c_rgb_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "charlcd_i2c_rgb_simpletest.py", line 18, in <module>
  File "adafruit_character_lcd/character_lcd_rgb_i2c.py", line 75, in __init__
  File "adafruit_mcp230xx/mcp23017.py", line 52, in __init__
  File "adafruit_mcp230xx/mcp23017.py", line 102, in iodir
  File "adafruit_mcp230xx/mcp230xx.py", line 49, in _write_u16le
  File "adafruit_mcp230xx/mcp230xx.py", line 49, in _write_u16le
OSError: [Errno 19] Unsupported operation
>>> 

it fails at the line
```
lcd = character_lcd.Character_LCD_RGB_I2C(i2c, lcd_columns, lcd_rows)
```
Note - the original issue was for a non-rgb test and it uses the mcp23008 and failed on a read attempt
https://discord.com/channels/327254708534116352/537365702651150357/806259603045351454

```

code run 
```
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""Simple test for I2C RGB character LCD shield kit"""
import time
import board
import busio
import adafruit_character_lcd.character_lcd_rgb_i2c as character_lcd

# Modify this if you have a different sized Character LCD
lcd_columns = 16
lcd_rows = 2

# Initialise I2C bus.
i2c = busio.I2C(board.GP1, board.GP0)

# Initialise the LCD class
lcd = character_lcd.Character_LCD_RGB_I2C(i2c, lcd_columns, lcd_rows)

lcd.clear()
# Set LCD color to red
lcd.color = [100, 0, 0]
time.sleep(1)
# Print two line message
lcd.message = "Hello\nCircuitPython"
# Wait 5s
time.sleep(5)
# Set LCD color to blue
lcd.color = [0, 100, 0]
time.sleep(1)
# Set LCD color to green
lcd.color = [0, 0, 100]
time.sleep(1)
# Set LCD color to purple
lcd.color = [50, 0, 50]
time.sleep(1)
lcd.clear()
# Print two line message right to left
lcd.text_direction = lcd.RIGHT_TO_LEFT
lcd.message = "Hello\nCircuitPython"
# Wait 5s
time.sleep(5)
# Return text direction to left to right
lcd.text_direction = lcd.LEFT_TO_RIGHT
# Display cursor
lcd.clear()
lcd.cursor = True
lcd.message = "Cursor! "
# Wait 5s
time.sleep(5)
# Display blinking cursor
lcd.clear()
lcd.blink = True
lcd.message = "Blinky Cursor!"
# Wait 5s
time.sleep(5)
lcd.blink = False
lcd.clear()
# Create message to scroll
scroll_msg = "<-- Scroll"
lcd.message = scroll_msg
# Scroll to the left
for i in range(len(scroll_msg)):
    time.sleep(0.5)
    lcd.move_left()
lcd.clear()
time.sleep(1)
lcd.message = "Going to sleep\nCya later!"
time.sleep(5)
# Turn off LCD backlights and clear text
lcd.color = [0, 0, 0]
lcd.clear()
```

jerryneedell, 2021-02-03:
Also ran against current tip of main -- same result
```
Adafruit CircuitPython 6.2.0-beta.1-68-gcd616f639 on 2021-02-03; Raspberry Pi Pico with rp2040
>>> import charlcd_i2c_rgb_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "charlcd_i2c_rgb_simpletest.py", line 18, in <module>
  File "adafruit_character_lcd/character_lcd_rgb_i2c.py", line 75, in __init__
  File "adafruit_mcp230xx/mcp23017.py", line 52, in __init__
  File "adafruit_mcp230xx/mcp23017.py", line 102, in iodir
  File "adafruit_mcp230xx/mcp230xx.py", line 49, in _write_u16le
  File "adafruit_mcp230xx/mcp230xx.py", line 49, in _write_u16le
OSError: [Errno 19] Unsupported operation
>>> 

```

tannewt, 2021-02-04:
Looks like we could be getting an error from pico-sdk: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/busio/I2C.c#L148-L157

LennartPiro, 2021-02-06:
I have the same issue when running the mcp230xx_simpletest.py with an MCP23017:

```mcp = MCP23017(i2c)
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
  File "adafruit_mcp230xx/mcp23017.py", line 52, in __init__
  File "adafruit_mcp230xx/mcp23017.py", line 102, in iodir
  File "adafruit_mcp230xx/mcp230xx.py", line 49, in _write_u16le
  File "adafruit_mcp230xx/mcp230xx.py", line 49, in _write_u16le
OSError: [Errno 19] Unsupported operation
```

on line
https://github.com/adafruit/Adafruit_CircuitPython_MCP230xx/blob/a6916350905dc1d3ad1ea47e8bbd32abc66169a8/examples/mcp230xx_simpletest.py#L25

dhalbert, 2021-02-08:
@jedgarpark is also seeing a similar with with SSD1306.

jedgarpark, 2021-02-09:
FWIW the problem went away for me when I included the I2C address in my code `display = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c, addr=0x3d)`

jerryneedell, 2021-02-09:
This has been Resolved for the LCD -- I had forgotten to copy the adafruit 74hc595 library to my Pico.
Once I copied this to, the charlie_i2c_rgb_simpletest runs ok.

dhalbert, 2021-02-09:
@jerryneedell What hw are you using? the I2C/SPI backpack uses an MCP23008 so I am not sure what combination of things you are using.

jerryneedell, 2021-02-09:
@dhalbert I'm using one of these https://www.adafruit.com/product/1109 - i have it connected to 5V GND, SDA,SCL

dhalbert, 2021-02-10:
@jerryneedell I tried this with the I2C/SPI backpack, which uses an MCP23008. I had to use 3.3v->5v level shifters to get it to work with a Metro M4, and then I moved it to an RPi Pico and it worked fine. I did not try it without the level shifters on the Pico in  the interest of not potentially frying the Pico.

If I use the Metro M4 connected straight to the backpack with the 5V pullups, it hangs up. I think it does not like the overvoltage. I have seen this before with the SAMD51

The PiLCDPlate (1109) you are using does not have any builtin pull-up resistors. The RPi's it's meant to work with already have pull-up resistors on their I2C lines. So your troubles may have to do with that. I also chatted with Redat on discord and their I2C LCD backup has 5v pullups.

jerryneedell, 2021-02-10:
@dhalbert Thanks, I did add 10K pull-ups -- and I think the single levels are since it it designed for a Pi. It only takes 5V input.
I'm really puzzled by the behavior on the Pico.  Alos puzzled by the guide reference to the adafruit_74hc595 library ... It does not seem to be used.

dhalbert, 2021-02-10:
The 74hc595 library is used only for SPI. The backpack has an MCP23008 for I2C and a 74HC595 for SPI use. I will clarify the Guide.

dhalbert, 2021-02-10:
Were your pullups to 3.3V or to 5V?

jerryneedell, 2021-02-10:
3.3v  -- and I just got it to work by removing the pull-ups and adding another I2C breakout to the setup to let it provide the pull-ups.  Checking my wiring, but it seems to be happy now...

jerryneedell, 2021-02-10:
Should I close this?  It is not clear to me what was wrong before, but I don't think this is a CP issue.

dhalbert, 2021-02-10:
Yes, let's. Redat is getting some level shifters and will try with those. Their LCD is arrangement is different.

closed 2021-02-10 by dhalbert