adafruit/circuitpython #4121: Add board.I2C(), board.SPI(), board.UART() for Raspberry Pi Pico board

dhalbert, 2021-02-03:
It would be nice to have default `board` `busio` devices on the Raspberry Pi Pico. The question is which pins to pick.

In the [Get Started with MicroPython on Raspberry Pi Pico book](https://hackspace.raspberrypi.org/books/micropython-pico), the I2C examples use GPIO0 and GPIO1:

---
[image]
SPI uses GPIO2, 3, and 4:

---
[image]
---
There is no UART example.

In the [Raspberry Pi MicroPython SDK](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf) book/datasheet, the default pins are different than above:
I2C:
[image]
SPI:
[image]
UART:
[image]

If we chose the first set, then the beginner book examples are easily translated.
If we chose the second set, then [sample MicroPython code](https://github.com/raspberrypi/pico-micropython-examples) is easier to translate.

Tagging @tannewt, @kattni, and @ladyada for comment.

jerryneedell, 2021-02-03:
Not a big deal either way. Users have to learn to read the chart eventually ;-)
I like the first option. I expect there will be many folks following the book at first then trying CP.
Would you put the default UART on GP7 and GP8?

ladyada, 2021-02-03:
i think we try to matchy with the examples in the book, and add to the FAQ

kattni, 2021-02-03:
How ubiquitous is the SDK within CircuitPython - which is to say, how likely is the typical user to be aware of the SDK info? Our goal is to make CircuitPython as simple as possible to use. As long as everything is documented, I think it's almost irrelevant what pins we use. That said, if we think the book is going to be/already is popular, we should follow those examples for simplicity. Otherwise, we should consider what the decision means in the long run - is the SDK going to be a longer-used reference than the book?

dhalbert, 2021-02-03:
I think the average CircuitPython user is not going to be looking at the MicroPython SDK document. The book is much easier to read and I can imagine it being a taking-off point. So using the first set makes sense. The main thing is to document it clearly.

kattni, 2021-02-03:
As we do not have a UART example yet, please put UART on GP8/GP9. It's a UART pair, and avoids I2C and SPI suggested above.

tannewt, 2021-02-04:
My vote would be to follow the book.

However, I'm not a huge fan of adding default busses when the board doesn't document them itself. At some point its on the board maker to do the designation.

dhalbert, 2021-02-05:
It might be a question to ask the RPi folks if they are thinking about adding default labels on the silk in the future.

tannewt, 2021-02-05:
We've brought it up with them previously but haven't heard of any concrete plans. I wouldn't expect them to designated default busses. Our main goal was to have labels similar to those on the bottom.

dhalbert, 2021-02-08:
Closing this after discussion in the weekly CircuitPython meeting of 2021-02-08, timestamp 1:26:18. We decided that trying to ameliorate board designs that fail to label default pins is beyond our purview. We can document the problem in the FAQ section in the RPI Pico guide. If and when the board silkscreen is updated to designate default bus pins, we can reopen this.

closed 2021-02-08 by dhalbert