adafruit/circuitpython #4129: Having the BitmapFont lib installed breaks building

FoamyGuy, 2021-02-04:
I have installed the bitmap font library via pip
`pip install adafruit-circuitpython-bitmap_font`

I use it with Blinka_Displayio on this computer. 

I've noticed that if I have that library installed in my python environment then I will get this error when attempting to build circuitpython:
```
$ make BOARD=nucleo_f767zi
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Unable to check whether maximum number of endpoints is respected
QSTR updated
Traceback (most recent call last):
  File "../../tools/gen_display_resources.py", line 59, in <module>
    f.load_glyphs(set(ord(c) for c in all_characters))
  File "/home/timc/.local/lib/python3.8/site-packages/adafruit_bitmap_font/bdf.py", line 224, in load_glyphs
    current_info["bitmap"][start + x] = bit
TypeError: 'BitmapStub' object does not support item assignment
make: *** [../../supervisor/supervisor.mk:225: build-nucleo_f767zi/autogen_display_resources.c] Error 1
make: *** Deleting file 'build-nucleo_f767zi/autogen_display_resources.c'
```
I can uninstall the lib with pip and then build normally.

tannewt, 2021-02-05:
Any idea where BitmapStub is coming from? We should replace it with something that works better.

FoamyGuy, 2021-02-05:
I think it's from here: 
https://github.com/adafruit/circuitpython/blob/c6a7b00030ffd8d7a9f566622fc12fcc3fd0e85b/tools/gen_display_resources.py#L27

It looks like this file does some path manipulation and import tricks to load up it's own version of bitmap_font. When I have the real one in the environment it's causing it to import the real one instead of this one internal to the circuitpython repo.

jepler, 2021-02-05:
```
sys.path.append("bitmap_font")
sys.path.append("../../tools/bitmap_font")
```
For the thing placed in the path to _override_, it should be placed earlier in the path, like `sys.path.insert(0, "...")`.

FoamyGuy, 2021-02-05:
That does fix it! Thank you @jepler. I can make a PR with this fix tonight or tomorrow.

closed 2021-02-08 by dhalbert

A-Spork, 2023-02-08:
Hey, i am still getting the same error
`$ make BOARD=lolin_s2_mini
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
mkdir -p build-lolin_s2_mini/genhdr
GEN build-lolin_s2_mini/genhdr/moduledefs.h
IDF_PATH=/tmp/circuitpython/ports/espressif/esp-idf cmake -S . -B build-lolin_s2_mini/esp-idf -DSDKCONFIG=build-lolin_s2_mini/esp-idf/sdkconfig -DSDKCONFIG_DEFAULTS="esp-idf-config/sdkconfig.defaults;esp-idf-config/sdkconfig-opt.defaults;esp-idf-config/sdkconfig-4MB.defaults;esp-idf-config/sdkconfig-esp32s2.defaults;boards/lolin_s2_mini/sdkconfig" -DCMAKE_TOOLCHAIN_FILE=/tmp/circuitpython/ports/espressif/esp-idf/tools/cmake/toolchain-esp32s2.cmake -DIDF_TARGET=esp32s2 -GNinja -DCIRCUITPY_ESP32_CAMERA=1
Including esp32-camera
COMPONENTS = esptool_py;soc;driver;log;main;esp-tls;mbedtls;mdns;esp_event;esp_adc_cal;esp_netif;esp_wifi;lwip;ulp;wpa_supplicant;freertos;bt;usb;esp32-camera
-- Found Git: /usr/bin/git (found version "2.34.1")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/adrian/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/adrian/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/adrian/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
-- Checking Python dependencies...
Python requirements from /tmp/circuitpython/ports/espressif/esp-idf/requirements.txt are satisfied.
-- Project sdkconfig file /tmp/circuitpython/ports/espressif/build-lolin_s2_mini/esp-idf/sdkconfig
Loading defaults file /tmp/circuitpython/ports/espressif/esp-idf-config/sdkconfig.defaults...
Loading defaults file /tmp/circuitpython/ports/espressif/esp-idf-config/sdkconfig-opt.defaults...
Loading defaults file /tmp/circuitpython/ports/espressif/esp-idf-config/sdkconfig-4MB.defaults...
Loading defaults file /tmp/circuitpython/ports/espressif/esp-idf-config/sdkconfig-esp32s2.defaults...
Loading defaults file /tmp/circuitpython/ports/espressif/boards/lolin_s2_mini/sdkconfig...
-- Found Python3: /home/adrian/.espressif/python_env/idf4.4_py3.10_env/bin/python3.10 (found version "3.10.6") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "circuitpython" version: 8.0.0-beta.6-168-g4d0af0ca3e-di
-- Adding linker script /tmp/circuitpython/ports/espressif/build-lolin_s2_mini/esp-idf/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_system/ld/esp32s2/sections.ld.in
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-time.ld
-- Adding linker script /tmp/circuitpython/ports/espressif/esp-idf/components/soc/esp32s2/ld/esp32s2.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt console cxx driver efuse esp-tls esp32-camera esp32s2 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hw_support esp_ipc esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py freertos hal heap ieee802154 log lwip mbedtls mdns newlib nghttp nvs_flash openthread partition_table pthread soc spi_flash tcpip_adapter ulp usb vfs wpa_supplicant xtensa
-- Component paths: /tmp/circuitpython/ports/espressif/esp-idf/components/app_trace /tmp/circuitpython/ports/espressif/esp-idf/components/app_update /tmp/circuitpython/ports/espressif/esp-idf/components/bootloader /tmp/circuitpython/ports/espressif/esp-idf/components/bootloader_support /tmp/circuitpython/ports/espressif/esp-idf/components/bt /tmp/circuitpython/ports/espressif/esp-idf/components/console /tmp/circuitpython/ports/espressif/esp-idf/components/cxx /tmp/circuitpython/ports/espressif/esp-idf/components/driver /tmp/circuitpython/ports/espressif/esp-idf/components/efuse /tmp/circuitpython/ports/espressif/esp-idf/components/esp-tls /tmp/circuitpython/ports/espressif/esp32-camera /tmp/circuitpython/ports/espressif/esp-idf/components/esp32s2 /tmp/circuitpython/ports/espressif/esp-idf/components/esp_adc_cal /tmp/circuitpython/ports/espre [... cut ...]
TypeError: 'BitmapStub' object does not support item assignment
make: *** [../../supervisor/supervisor.mk:235: build-lolin_s2_mini/autogen_display_resources-en_US.c] Error 1
make: *** Deleting file 'build-lolin_s2_mini/autogen_display_resources-en_US.c'`
when building different espressif boards. Even when adafruit-circuitpython-bitmap_font is not installed. I allready tried installing/ uninstalling adafruit-circuitpython-bitmap_font and replacing @jepler ´s fix with the old version, but it did not work yet.

FoamyGuy, 2023-02-09:
@A-Spork do you have the same error if you attempt to build something other than esspressif port? 

Are you building from current `main` branch or something older / different?

A-Spork, 2023-02-09:
I am building from the current main. The circuitplayground_express used in the example does work with and without adafruit-circuitpython-bitmap_font installed.

jepler, 2023-02-09:
(I edited a comment to use the triple-backticks method of formatting code so that line breaks and white space were preserved)

jepler, 2023-02-09:
Closed issues are not a good place for support questions.

It *looks like* the content you have in the file tools/bitmap_font/adafruit_bitmap_font/bdf.py does not match what we have in git. For instance, your error pertains to line 229 of that file, while the current version contains only 109 lines.

You need to investigate why your system has different content in the tools/bitmap_font/adafruit_bitmap_font/bdf.py file and resolve it.

```
$ git submodule status tools/bitmap_font/
 62dd78abdd0b823824fe15d1bab0611246145c23 tools/bitmap_font (1.0.0~2)
```

tannewt, 2023-02-09:
> Closed issues are not a good place for support questions.

Its much better to open a new issue and reference this issue from there.

PR #4140 (2021-02-08): fix bitmap_font path in display resource gen
This change allows you to successfully build circuitpython even if you have the bitmap_font library installed in your python environment.

I'm guessing that this script is involved in making `terminalio.FONT` but am not certain. I tested the resulting build on a PyPortal with a script that draws text using `terminalio.FONT` and it does appear to work normally as far as I can tell.

resolves: #4129