adafruit/circuitpython #4132: Requests fail if too many DNS servers presented via DHCP

funjon, 2021-02-05:
Not sure if this is a CircuitPython issue, or an issue with the ESP network code. My home network issues 4 DNS servers (two comcast-issued DNS servers, and 8.8.8.8/1.1.1.1 as backups) via DHCP. When my MagTag gets all 4, requests with FQDN's fail:

`Connecting to dragonfly-manor
Connected to dragonfly-manor!
My IP address is 10.225.0.6
Ping google.com: 38.999985 ms
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
Traceback (most recent call last):
  File "code.py", line 41, in <module>
  File "adafruit_requests.py", line 595, in get
  File "adafruit_requests.py", line 559, in request
  File "adafruit_requests.py", line 419, in _get_socket
OSError: 0
`

If I make the request via IP, it works:
`
$ host wifitest.adafruit.com
wifitest.adafruit.com has address 104.236.193.178
`

`
Connected to dragonfly-manor!
My IP address is 10.225.0.6
Ping google.com: 33.999984 ms
Fetching text from http://104.236.193.178/testwifi/index.html
----------------------------------------
This is a test of Adafruit WiFi!
If you can read this, its working :)

----------------------------------------
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
Traceback (most recent call last):
  File "code.py", line 48, in <module>
  File "adafruit_requests.py", line 595, in get
  File "adafruit_requests.py", line 559, in request
  File "adafruit_requests.py", line 419, in _get_socket
OSError: 0
`

And if I limit it to two DNS servers, everything appears to be working fine:

`
Connecting to dragonfly-manor
Connected to dragonfly-manor!
My IP address is 10.225.0.6
Ping google.com: 62.999988 ms
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
----------------------------------------
This is a test of Adafruit WiFi!
If you can read this, its working :)

----------------------------------------
`

anecdata, 2021-02-05:
What board are you using, and what versions of CircuitPython and Requests?

funjon, 2021-02-07:
I am using a recent MagTag board, and the 6.1.0 release of CircuitPython (and whatever adafruit_requests.mpy comes with it)

hierophect, 2021-02-10:
This could be simply that you're running out of sockets, since any TLS sockets you use will actually reserve 2, and we only have 4 total right now for memory reasons. That said, if you were, it should raise a message specific to that and not the placeholder `OSError: 0`

anecdata, 2021-02-10:
Can you try with the "Absolute Newest" CircuitPython https://circuitpython.org/board/adafruit_magtag_2.9_grayscale/ and latest release of Requests? This may be a distinct DNS issue, but there has been a lot of rework and issues closed since 6.1.

funjon, 2021-02-12:
With the latest versions of code, it looks like it's working now. Thanks for looking!

dhalbert, 2023-10-28:
Reported as working, so closing.

closed 2023-10-28 by dhalbert