adafruit/circuitpython #4133: Revamp status LED signaling

dhalbert, 2021-02-05:
@tannewt discussed revisiting the multi-colored status LED blinking that CIrcuitPython does to indicate errors. I think the ideas were to:
- simplify the indications
- do not repeat them indefinitely

@tannewt Please revise this post or comment as necessary.

tannewt, 2021-02-05:
Yup! I think we only want to use the status LED on power up (not every reset) and on error. The main goal is to save power.

Dygear, 2021-04-19:
Watching the live stream from [this week's deep dive](https://youtu.be/Q4sUzNMQA0U?t=4871) and thinking about colors. I was thinking something like ...

- Red for Compile Time Errors (Blink Morse Code "SOS" x2)
- Yellow for user land thrown and uncaught Exceptions. (Blink Morse Code "Error" x2)
- Purple for when it's first running the code. (Blink Morse Code "Run")
- Green for when it gets to code completion. (Blink Morse Code "Done").

SOS = `... --- ...`
Error = `. .-. .-. --- .-.`
Run = `.-. ..- -.`
Done = `-.. --- -. .`
Help = `.... . .-.. .--.`

So why the color and then morse code? Some people are color blind, this allows them to get some information from this LED as well. I am not color blind myself, but designing UI / UX for governmental type organizations, these types of things are a requirement due to the [ADA](https://www.dol.gov/general/topic/disability/ada) -- As well as ya know just being nice.

tannewt, 2021-04-19:
> [Edit] Oh, and like a minute after I paused to write this you said you didn't want morse code and you're going into much more detail about what you have planned. I would really wait until the end of the video before I make comments. LOL. Sorry.

No worries! Thanks for the thinking on this.

hierophect, 2021-04-19:
This is kind of a trivial note but how will this activate with regards to deep sleep? Currently, while waiting for USB to enumerate, Deep Sleep iterates constantly over the `tick_rgb_status_animation` call in main. There is also the general difference between true and fake deep sleep in terms of how they trigger a restart, and the fact that chips like the RP2040 which do not reset their own flash may always use the fake deep sleep method. I have no real preference on the actual specifics of these points, just bringing them up in case you feel any of them might be important.

tannewt, 2021-04-19:
My plan is to get rid of the tick function and make the loop in main more like the time.sleep loop. So, between changing flashes it would light sleep.

PR #4743 (2021-05-21): Simplify the status LED to save power
This also removes the need to pin share because we don't use the
status LED while user code is running.

The status flashes fallback to the HW_STATUS LED if no RGB LED is
present. Each status has a unique blink pattern as well.

One caveat is the REPL state. In order to not pin share, we set the
RGB color once. PWM and single color will be shutoff immediately but
DotStars and NeoPixels will hold the color until the user overrides
it.

Fixes #4133
commit 1a0b4193 (2021-05-21): Simplify the status LED to save power

This also removes the need to pin share because we don't use the
status LED while user code is running.

The status flashes fallback to the HW_STATUS LED if no RGB LED is
present. Each status has a unique blink pattern as well.

One caveat is the REPL state. In order to not pin share, we set the
RGB color once. PWM and single color will be shutoff immediately but
DotStars and NeoPixels will hold the color until the user overrides
it.

Fixes #4133
closed 2021-05-21 by tannewt