adafruit/circuitpython #4142: MagTag: board.SPI() returns "SCK in use"

anecdata, 2021-02-06:
Latest S3:
```
Adafruit CircuitPython 6.2.0-beta.1-115-ge7438c481 on 2021-02-05; Adafruit MagTag with ESP32S2
>>> import board
>>> spi = board.SPI()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SCK in use
```

Neradoc, 2021-02-06:
Testing different builds on S3, I see the error existed on January 1, disappears after #3930 "UMFEATHERS2 - implement use of DotStar for status led", and consequently comes back after #4013 which reverts 3930.

The original PR included the following change in `ports/esp32s2/common-hal/microcontroller/Pin.c`

```C
bool pin_number_is_free(gpio_num_t pin_number)
(...)
-    return (never_reset_pins[offset] & mask) == 0 && (in_use[offset] & mask) == 0;
+    return (in_use[offset] & mask) == 0;
```

I quote @hierophect :
> I don't know why pin_is_free is checking the never_reset tables, that is nonstandard port behavior and is not shared for the STM32, NRF or SAMD ports. It's not something we ever formally hashed out though. I'd recommend the (never_reset_pins[offset] & mask) part of that line should actually be deleted in Microcontroller unless @tannewt has objections - I think cases like were one of the original reasons we opted to keep claimed_pins and never_reset_pins separate, IIRC.

hierophect, 2021-02-15:
So if I understand correctly, all that needs to be done here is to recreate that fix (deleting the bad never_reset check) and submit it as its own PR? This is the risk of folding in important fixes into semi-unrelated PRs I suppose.

anecdata, 2021-02-15:
Looks good to me:
```
Adafruit CircuitPython 6.2.0-beta.2-16-gbc6db350e on 2021-02-15; Adafruit MagTag with ESP32S2
>>> import board
>>> spi = board.SPI()
>>> spi.frequency
250000
```

anecdata, 2021-02-15:
Fixed by #4205

anecdata, 2021-02-17:
This fix seems to change refresh behavior:
```
import time
import board
import supervisor
import microcontroller
from adafruit_magtag.magtag import MagTag

def show():
    for _ in range(0, 5):
        print(" ", _, "\r", end="")
        magtag.set_text(_)
        time.sleep(magtag.graphics.display.time_to_refresh + 1)
        magtag.graphics.display.refresh()

time.sleep(5)  # time to get serial console connected

magtag = MagTag()

magtag.add_text(
    text_position=(
        magtag.graphics.display.width // 2,
        magtag.graphics.display.height // 2,
    ),
    text_scale=4,
)
# this is fine
print("Before SPI")
show()

spi = board.SPI()

# now this is not fine
print("After SPI")
show()

# this doesn't fix it
# supervisor.reload()

# this does
microcontroller.reset()
```

DavePutz, 2021-02-28:
It looks like the MagTag is using the default SPI pins (GPIO36 for SCK and  GPIO35 for MOSI) for the E-ink display. In order to use SPI for some other device you might need to use busio.SPI instead of board.SPI.

deshipu, 2021-02-28:
Maybe it would make sense to move the default SPI to other pins.

Neradoc, 2021-02-28:
Why does the MagTag even have a board.SPI ? What physical pins is that for ?
Is it to address the display directly ? Does it matter with board.DISPLAY being there ?

anecdata, 2021-03-01:
It's mainly a code portability thing. I should think the singleton shouldn't error.

dhalbert, 2021-05-01:
I propose that we drop `board.SPI()`, since it has no utility other than to talk to the display. I will also rename the SPI pins to `EPD_MOSI`, etc. The fact that they were called plain `MOSI`, etc., is an historical artifact, I believe, based on the original schematic.

anecdata, 2021-05-01:
Addressed by #4695 ...closing.

PR #4695 (2021-05-01): MagTag: Remove board.SPI(); rename display-only pins
On the MagTag, `board.SPI()` can only be used for the display; the pins are not broken out elsewhere. Using `board.SPI()` gives a "pin in use" error.

- Remove `board.SPI()`.
- Rename `board.MOSI`, `.MISO`, `.SCK` to `.EPD_MOSI`, etc.
- Remove `board.CS`; it already has an alias `EPD_CS`.

Fixes #4142.
closed 2021-05-01 by anecdata