adafruit/circuitpython #4143: RP2040 PWMIO module throws All timers for this pin are in use when defining two or more pins

jposada202020, 2021-02-06:
When using the following code, I am getting the error that all the timers are used for pinblue(GP19)

    import pwmio
    import board
    
    
    pinred = pwmio.PWMOut(board.GP17, frequency=5000, duty_cycle=0)
    pingreen = pwmio.PWMOut(board.GP18, frequency=5000, duty_cycle=0)
    pinblue = pwmio.PWMOut(board.GP19, frequency=5000, duty_cycle=0)

code.py output:
Traceback (most recent call last):
  File "code.py", line 7, in <module> 
ValueError: All timers for this pin are in use

On the REPL this was reproduced

Adafruit CircuitPython 6.2.0-beta.1 on 2021-01-27; Raspberry Pi Pico with rp2040
>>> import pwmio
>>> import board
>>> pinred = pwmio.PWMOut(board.GP17, frequency=5000, duty_cycle=0)
>>> pingreen = pwmio.PWMOut(board.GP18, frequency=5000, duty_cycle=0)
>>> pinblue = pwmio.PWMOut(board.GP19, frequency=5000, duty_cycle=0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: All timers for this pin are in use

Inversing the order of the pin definition will throw the same error, so I am assuming this is related with the pwmio definitions
Adafruit CircuitPython 6.2.0-beta.1 on 2021-01-27; Raspberry Pi Pico with rp2040
>>> import pwmio
>>> import board
>>> pinred = pwmio.PWMOut(board.GP17, frequency=5000, duty_cycle=0)
>>> pinblue = pwmio.PWMOut(board.GP19, frequency=5000, duty_cycle=0)
>>> pingreen = pwmio.PWMOut(board.GP18, frequency=5000, duty_cycle=0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: All timers for this pin are in use
>>> 

This code will pass

    import pwmio
    import board
    ping1 = pwmio.PWMOut(board.GP21, frequency=700)
    ping2 = pwmio.PWMOut(board.GP22, frequency=5000)
    ping4 = pwmio.PWMOut(board.GP11, frequency=400)
    ping5 = pwmio.PWMOut(board.GP16, frequency=400)
  
this will fail

    import pwmio
    import board
    ping1 = pwmio.PWMOut(board.GP21, frequency=700)
    ping2 = pwmio.PWMOut(board.GP22, frequency=5000)
    ping4 = pwmio.PWMOut(board.GP11, frequency=400)
    ping5 = pwmio.PWMOut(board.GP16, frequency=400)
    ping6 = pwmio.PWMOut(board.GP5, frequency=400)

I am wondering if when the mask is defined, in the code, we are missing something. I cannot test that part of the code. However, if you select any pins in the same (8 2 Channel Slice) the code will fail. If we take the table 526 in the RP2040 datasheet, we could predict more or less which pins will fail.  For example, for PIN 2, associated pins in the (8 2 Channel) will be 3, 8 and 9 and if we try in the following code all will have the same error. 

    import pwmio
    import board
    ping1 = pwmio.PWMOut(board.GP2, frequency=550)
    ping2 = pwmio.PWMOut(board.GP19, frequency=550) 

So I am not sure if this is related with the mask or the slice definition

tannewt, 2021-02-09:
@DavePutz Do you want to take a look at this and find my bug?

timonsku, 2021-02-09:
hitting the same issue when using just two pwm's (16, 17)

DavePutz, 2021-02-09:
@tannewt - I'll take a look at it.

tannewt, 2021-02-09:
Thanks @DavePutz!

DavePutz, 2021-02-09:
@tannewt - We are currently checking whether a PWM channel is in use and throwing an error if it is. That is why the reported error it hit when GPIOs 21 and 5 are both used - they share channel 2B. However, the datasheet says "The same PWM output can be selected on two GPIO pins; the same signal will appear on each GPIO.". So, should we just take out the check for channel re-use?

tannewt, 2021-02-09:
@DavePutz No, lets not allow outputting the same channel to multiple pins because the api implies their duty cycle would be independent.

DavePutz, 2021-02-09:
@tannewt - Ok, I'll submit a PR for another issue causing the error (we throw the same error for 4 different conditions :-(  and leave the channel check in.

jedgarpark, 2021-02-10:
Finding this issue here just saved me a lot of heartache :) I didn't pay attention two the "two pins per PWM channel" thing until seeing this.

timonsku, 2021-02-10:
oh so this is fixed? I thought the PWM channels could be freely mapped. Where did you find that table with the channel pairs?

jfurcean, 2021-02-10:
Table 526 on page 536 of the [data sheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf). Also in Table 290 on page 252.

tannewt, 2021-02-10:
This was fixed by #4165

closed 2021-02-10 by tannewt

Bhavithiran97, 2021-02-11:
I am still getting "ValueError: All timers for this pin are in use" when using GP18 and GP19 as PWMOut.

DavePutz, 2021-02-11:
With the patch in, this script is working for me:
import pwmio
import board
pinred = pwmio.PWMOut(board.GP17, frequency=5000, duty_cycle=0)
pingreen = pwmio.PWMOut(board.GP18, frequency=5000, duty_cycle=0)
pinblue = pwmio.PWMOut(board.GP19, frequency=5000, duty_cycle=0)
print(pinred,pingreen,pinblue)

Note that if you choose GPIO pins that use the same channel (such as 5 and
21, or any pair
that differ by 16) you will still get the ValueError.

Bhavithiran97, 2021-02-11:
What do you mean by patch in ?

d-c-d, 2021-02-11:
I think they mean “with the  proposed  changes incorporated”

dhalbert, 2021-02-11:
Did you test again with 6.1.0-beta.2? It incorporates the #4165 fix.

Bhavithiran97, 2021-02-15:
`import pwmio
import board
ping1 = pwmio.PWMOut(board.GP18, variable_frequency=True)
ping2 = pwmio.PWMOut(board.GP19, variable_frequency=True)`

Getting "ValueError: All timers for this pin are in use".

I want to change the frequency in a function, both GP18 and GP19 to same frequency. How can I do that ?

timonsku, 2021-02-15:
@Bhavithiran97 you can't, check out the table here: https://github.com/adafruit/circuitpython/issues/4143#issuecomment-776388367
Some pins are paired.  While you can adjust the duty cycle you can not adjust the frequency. If you need a different frequency on each pin you need to choose a different set of pins that don't share their PWM bank.
What this patch fixes is that you can still use them both but not if you construct them both with `variable_frequency=True` as you can not change their frequency independently in this configuration. If you were to use GP20 instead of GP19 f.e. this would work again.

tannewt, 2021-02-16:
> I want to change the frequency in a function, both GP18 and GP19 to same frequency. How can I do that ?

The CircuitPython API doesn't allow for two PWM outputs to share a variable frequency. Instead, variable frequency requires separate underlying hardware. On the RP2040 that means separate PWM slices.

Bhavithiran97, 2021-02-18:
`import machine
import utime

C4  = 262
D4  = 294
E4  = 330

tone = [C4,D4,E4]

audio_left = machine.PWM(machine.Pin(18))
audio_right = machine.PWM(machine.Pin(19))

for i in tone:
    audio_left.freq(i)
    audio_right.freq(i)
    audio_left.duty_u16(19660)
    audio_right.duty_u16(19600) #30%
    utime.sleep(0.15)
    audio_left.duty_u16(0)
    audio_right.duty_u16(0)
`

I can use change both GP18 and GP19 frequency in MicroPython. I can't do in  in CircuitPython ?

Bhavithiran97, 2021-02-18:
```
import machine
import utime

C4  = 262
D4  = 294
E4  = 330

tone = [C4,D4,E4]

audio_left = machine.PWM(machine.Pin(18))
audio_right = machine.PWM(machine.Pin(19))

for i in tone:
    audio_left.freq(i)
    audio_right.freq(i)
    audio_left.duty_u16(19660)
    audio_right.duty_u16(19600) #30%
    utime.sleep(0.15)
    audio_left.duty_u16(0)
    audio_right.duty_u16(0)

```
I can change both frequecy for GP18 and GP19 in MicroPython. I can't do it in CircuitPython ?

tannewt, 2021-02-18:
> I can change both frequecy for GP18 and GP19 in MicroPython. I can't do it in CircuitPython ?

No. frequency is dictated by the TOP value and clock divider which are shared for the two channels output from a PWM slice. CircuitPython prevents you from changing these values from a single output because it could lead to unexpectedly changing the frequency of the other channel. MicroPython doesn't check this for you. Therefore, in CircuitPython you need to use pins with separate slices for each output.

(For audio playback check out `audiopwmio` that is in the "absolute latest" builds currently.)

tannewt, 2024-01-18:
> Running Circuitpython 8.2.9 on Itsybitsy M4:

Please file a new issue. This issue is about the RP2040. Not the SAMD51 "M4".

dhalbert, 2024-01-18:
New issue created: #8811.