adafruit/circuitpython #4147: ESP32S2: unable to access I2C after soft reboot

jerryneedell, 2021-02-06:
While testing #4146  I ran into this behavior on a MagTag.
It is also reproducible on a metro_esp32s2
but 
it does not occur on a PyPortal (samd51)

the issue is that I can access the I2C device after a Power Cycle or Hard Reset but it fails after a soft reboot on esp32s2 boards.

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.1-14-gea4a12005-dirty on 2021-02-06; Adafruit MagTag with ESP32S2
>>> import board,adafruit_si7021
>>> sht = adafruit_si7021.SI7021(board.I2C())
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.1-14-gea4a12005-dirty on 2021-02-06; Adafruit MagTag with ESP32S2
>>> 
>>> import board,adafruit_si7021
>>> sht = adafruit_si7021.SI7021(board.I2C())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_si7021.py", line 99, in __init__
ValueError: No I2C device at address: 40
>>> 
```

jerryneedell, 2021-02-06:
May be related to #3894

anecdata, 2021-02-06:
May also be related to #4046

m4tk, 2021-02-07:
Could reproduce it on a LilyGo T8 ESP32-S2 and a cheapy BME280. But it's occurring only sporadic (maybe in 1 of 10 soft reboots in a row).

skieast, 2021-02-08:
I crash my Cucumber when doing this. I had to do some I2C changes as the Cucumber has no pullups on its internal I2C bus (not great).  Moving some of the constructor bits around seems to get rid of the crash but then nothing is found on doing a scan.  I will keep looking.  I am also wondering if espressif has put changes/fixes into the idf since the commit that we (CPY) use.   I did try building wit version 4.2 of the idf, nothing really changed.

jerryneedell, 2021-02-11:
This is pretty much a show stopper for using es32s2 with I2C -- Should this be assigned to a milestone? I think it is fairly urgent.

jerryneedell, 2021-02-11:
hmm -- This issue doe snot occurs with an U expected Maker feathers2 and a bme680 I2C sensor.
It is able to run after reboot. 

No idea if it is relevant, but I do not have the UF2 Bootloader on the umfeathers2.

jerryneedell, 2021-02-11:
Ah HA! This issue doe not occur on the Metro-esp32s2 if I do not use the UF2 Bootloader.... the plot thickens....

skieast, 2021-02-11:
I was wondering when this stopped working.  
Works with
    Adafruit CircuitPython 6.1.0-beta.2-7-g22ad76bb4-dirty on 2020-12-07; FeatherS2 with ESP32S2

Doesnt Work (built today from main)
    Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; FeatherS2 with ESP32S2

hierophect, 2021-02-15:
I'm consolidating this with a number of other issues that all appear to be the same thing. Please continue discussion of this issue in #4046.

closed 2021-02-15 by hierophect