adafruit/circuitpython #4152: Empty packets processed by socket/recv_into on ESP32 return OSError EBADF

benetherington, 2021-02-08:
I've been banging my head against this for a while, so apologies if this isn't clear!

Background: I'm trying to connect my Magtag to a Mosquitto server.  I'm able to subscribe, publish, and receive published messages. However, I'm encountering errors when `loop()`ing when there are no messages to receive. Note that if you want to reproduce this, you'll have to use an IP for secrets['mosquitto_broker'], not a .local address because #4059.

When I run the following code, I get `OSError` raised, with `[Errno 9] EBADF`. This is raised from `adafruit_minimqtt`/`_sock_exact_recv`, line 909, in `self._sock.recv_into(rc, bufsize)`.

```
from time import monotonic, sleep
import ssl
import socketpool
import wifi
import ipaddress
import adafruit_minimqtt.adafruit_minimqtt as MQTT

from secrets import secrets

wifi.radio.connect(secrets["ssid"], secrets["password"])
print("Connected to WIFI")
pool = socketpool.SocketPool(wifi.radio)

mqtt_client = MQTT.MQTT(
  broker=secrets['mosquitto_broker'],
  port=1883,
  username=secrets['mosquitto_username'],
  password=secrets['mosquitto_password'],
  socket_pool=socketpool.SocketPool(wifi.radio)
)

def connected(mqtt_client, userdata, flags, rc):
  print("Connected to MQTT Broker!")
  print("Flags: {0}\n RC: {1}".format(flags, rc))
def subscribe(client, userdata, topic, granted_qos):
  print("Subscribed to {0} with QOS level {1}".format(topic, granted_qos))
def unsubscribe(client, userdata, topic, pid):
  print("Unsubscribed from {0} with PID {1}".format(topic, pid))
def disconnected(client):
  print("Disconnected from Adafruit IO!")
def message(client, feed_id, payload):
  print("Feed {0} received new value: {1}".format(feed_id, payload))

mqtt_client.on_connect = connected
mqtt_client.on_disconnect = disconnected
mqtt_client.on_subscribe = subscribe
mqtt_client.on_unsubscribe = unsubscribe
mqtt_client.on_message = message

mqtt_client.connect()
import adafruit_logging as logging
mqtt_client.enable_logger(logging, 10)
mqtt_client.subscribe('magtag/status')

# two _socket.send()s in a row
mqtt_client.ping()
mqtt_client.ping() # no problem here!
# two _socket.recv_into() in a row with no data fails
mqtt_client.loop() # socket closes
try:
  mqtt_client.loop()
except OSError:
  pass # raises OSError(9) EBADF
# two _socket.recv_into() in a row, but with data to fetch
mqtt_client.reconnect()
"publish sent to magtag/status from another machine"
mqtt_client.loop() # incoming message processed as expected
mqtt_client.loop() # socket closes, no exceptions
mqtt_client.ping() # _socket.send() still works!
try:
  mqtt_client.loop()
except OSError:
  pass # raises OSError(9) EBADF
try:
  mqtt_client.ping()
except OSError:
  pass # also raises OSError(9) EBADF. _socket.send() no longer works?
```

I don't know if this is helpful, but passing a buffer size of 0 to `recv_into` in `_sock_exact_recv` during `loop()` calls doesn't trigger this issue, but obviously it also results in unread messages piling up until the next `keep_alive` period ends and `ping` calls `_sock_exact_recv` with a non-zero buffer.

I'm sure there is more info I could include, or debugging steps I could take, but at this point I'm pushing my knowledge, and I can't see what they are. I'd be more than happy to do some more work on this if someone can lend a second pair of eyes and point me in the right direction!

hierophect, 2021-02-10:
@brentru I'm less familiar with the mqtt parts of this. EBADF is most likely to occur when a read or write operation is attempted on a socket that has been closed. Are there extra steps that should be taken here to ensure that valid sockets are being used in the loop?

brentru, 2021-02-11:
@hierophect  

> Are there extra steps that should be taken here to ensure that valid sockets are being used in the loop?

I'm not sure if the socket is being closed by the broker (MQTT terminology for "server") or the client (device)...I can try this locally with a Mosquitto broker.

benetherington, 2021-02-15:
The socket is being closed by the client according to my Mosquitto logs. `Client cpy3438 closed its connection.`