adafruit/circuitpython #4153: MCB1700 board support

EmergReanimator, 2021-02-08:
There are tons of devices based on NXP LPC17xx MCU around.
It would be great to provide the python environment for these boards.
One real-life example is production testing.

EmergReanimator, 2021-02-08:
Proposed labels:
good first issue
board
enhancement
nxp
mcb1700
lpc17xx

dhalbert, 2021-02-08:
We welcome ports to additional chip families. It's a lot of work, primarily due to supporting the clock architecture and peripherals of a particular chip family. Sometimes they resemble other chip families from the same manufacturer, sometimes not.

It sounds like you want to support existing boards, rather than a new design. The LPC17xx family, as you know is Cortex M3, which is not used in new designs that I have seen. The max RAM on the family is 96kB, which is adequate, but not great. Those chips are >$6 for 10k qty, which is noticeably higher than other comparable chips people have been using.

We have been concentrating more on the NXP i.MX RT10xx line. The LPC522 series is also interesting because its chips can have a substantial amount of RAM.

EmergReanimator, 2021-02-08:
Thanks for your quick response and hints!

Let me give you some background information of given issue. 
There are number of projects based on this MCU in our company. It will be superseded by SAME51 in next projects however. The life cycle of our products is at least 10 years. During this time we have to manufacture and test those boards.
I am going to provide low level access to MCU and peripherals to our HW developers and testers for existing devices.
Using circuitpython they will be able to reuse the python scripts on SAME51 as well at least partly.
Thus LPC17xx support is required for smooth transition to SAME5x.

I made some circuitpython evaluation in advance using SAME54 Xplained Pro board. So I am more or less aware of LPC17xx bottlenecks. Besides that I have experience of programming LPC17xx in different flavours. The support will be limited to rudimentary _busio_ features like I2C, SPI or CAN interfaces. The only interaction with circuitpython should occur via python shell over UART.

Wish me a good luck!

EmergReanimator, 2021-02-08:
> The LPC522 series is also interesting because its chips can have a substantial amount of RAM.

I believe you referer to LPC552x Cortex-M33, right?

> We have been concentrating more on the NXP i.MX RT10xx line. 

Is any code available publically already?

dhalbert, 2021-02-08:
Right LPC552, typo, sorry. Our i.MX port is in `circuitpython/ports/mimxrt10xx`.

dhalbert, 2021-02-08:
Thanks for the detailed explanation. What is the RAM and flash of the chip used on your existing board(s)? There are some lower limits below which it's very difficult to do any kind of build.

EmergReanimator, 2021-02-08:
> What is the RAM and flash of the chip used on your existing board(s)? There are some lower limits below which it's very difficult to do any kind of build.

RAM: 64 - 96 KiB
Flash: ~400 KiB

dhalbert, 2021-02-08:
That should be fine. I looked briefly at your repo and you have defined a port-specific `main.c` and done some other things that might be leftover MicroPython implementation. Take a look at a clean port like `atmel-samd`, `stm32`, or `nrf``.

tannewt, 2021-02-09:
My first rp2040 commit is also a good place to start: https://github.com/adafruit/circuitpython/commit/733094aead6e84b42da54655b2b0a5d0dbd1a503#diff-9a32e21d62fdae42eb9135c5c5247bb84aa3de605f4e64bb118d926cc591521e

TinyUSB support is a pre-requisite as well. If you have questions, feel free to ask them on the [Adafruit Discord](https://adafru.it/discord) in the #circuitpython channel.

commit e1ebc37d (2021-06-04): Merge pull request #4858 from EmergReanimator/main

resolves #4153: Fixed build issue when CIRCUITPY_USB is off
commit f2e3231c (2021-06-04): resolves #4153: Fixed build issue when CIRCUITPY_USB is off