adafruit/circuitpython #4156: MagTag randomly wiping itself

KTibow, 2021-02-08:
I recently set up an Adafruit MagTag. The code I'm running can be found at https://github.com/KTibow/fridge. It seems like it randomly resets to what it would look like if I cleared the flash, and installed CircuitPython. I have the same problem on 6.10 and the latest beta.
[image]

dhalbert, 2021-02-08:
What is in `code.py` when this happens. Is it `print("Hello, world!\n")`?

The way you are starting `code.py` from `boot.py` isn't necessary. `code.py` will run after `boot.py`, so you should be able to just omit these lines:
```python
# Finally code
exec(open("/code.py").read())
main()
```

A different thing you may want to do is to reset the board entirely if you've loaded a new `code.py`, by doing `microcontroller.reset()`. But try just omitting the lines above first

It is possible that there's a bug that's not completing the write to `code.py` even though it's flushed and closed. Since this is intermmittent, how often would you say that it happens.

KTibow, 2021-02-08:
> What is in code.py when this happens. Is it print("Hello, world!\n")?

Yes.

>The way you are starting code.py from boot.py isn't necessary. code.py will run after boot.py, so you should be able to just omit these lines:

I need to keep the context of the WiFi radio and the socket pool; in my experience, it gives me an issue if I didn't.

> A different thing you may want to do is to reset the board entirely if you've loaded a new code.py, by doing microcontroller.reset(). But try just omitting the lines above first

Huh. As long as I could tell the difference between that, and a user-initiated reset, then it would work.

> It is possible that there's a bug that's not completing the write to code.py even though it's flushed and closed. Since this is intermmittent, how often would you say that it happens.

It's probably not a problem with that. I put the MagTag on my fridge last night, and it was working fine. But when I woke up, without any interaction, it was reset with a blank screen. When I turned it off and on again, it displayed Hello World.

dhalbert, 2021-02-08:
If you check for the button push first, then you can do all the wifi setup conditionally. Then after writing code.py, you would reset, and you don't care when you restart about the previous state of the world.

KTibow, 2021-02-08:
I think I understand. I'll try to get to coding now.

KTibow, 2021-02-11:
The same thing happened last night. Do you think it has something to do with low battery levels?

dhalbert, 2021-02-11:
We saw something similar a long time ago with SAMD boards. They would reboot at too low a voltage level: the voltage was enough to run the CPU but the first attempt at trying to read the flash chip failed. So the code assumed the flash filesystem was defective and reformatted it. We ended up changing the bootloader to wait longer for the voltage to stabilize.

The latest 6.1.0-beta.2 has a fix so that waking from deep sleep will not mistakenly think there was a brownout. I'm not sure that would help here but you could try it. Check your battery voltage and charge it a bit but not all the way to speed up the testing.

KTibow, 2021-02-13:
I tried installing it, however I left it on overnight, and the same thing happened. Is there a way for me to manually know what the battery level is so I can just exit the program?

KTibow, 2021-02-13:
I managed to find how to access the battery (`magtag.peripherals.battery`).

> the voltage was enough to run the CPU but the first attempt at trying to read the flash chip failed

About at what voltage did that happen?

dhalbert, 2021-02-13:
The flash chips we use typical have a minimum operating voltage of 2.6v. The exact chip used in the ESP32-S2 module is not specific, but it might be similar. The datasheet for the module says a recommended minimum operating voltage is 3.0v. So it would be interesting to know if your voltage is dipping below either of those.

We could add code to CircuitPython that refuses to run below a certain voltage. It could also wait longer on startup for the voltage to stabilize.

KTibow, 2021-02-13:
Got it. I've added a simple
```
    if magtag.peripherals.battery < 3.3:
        break
```
I'll see what happens tonight.

KTibow, 2021-02-14:
That simple thing didn't work. I'll try raising the cap. Something I also noticed is when I wake up, the display is always completely black.

dhalbert, 2021-03-31:
Have you resolved this problem? Please try the latest beta. If you are connected to USB, then it will not go into real deep sleep.

closed 2021-03-31 by dhalbert

KTibow, 2021-03-31:
Right now, I've implemented deep sleep, and added an external battery. I haven't experienced the issue since.