adafruit/circuitpython #4158: "Large" file transfer to CIRCUITPY

gc-elec, 2021-02-08:
Hello,

As you know, I built a few custom board prototypes using SPI Flash GD25Q64C.
CircuitPython now works great, but I noticed a problem with file transfer to the CIRCUITPY drive.

When I try to send a "large" file (87 kB wave file), the transfer stays stuck at 0% and never complete.
[image]
A 23 kB files is ok
A 63 kB file stayed stuck for a while at 33% and eventually finished after around 30 seconds.

I don't have any problem with small files (a few kB). Only larger files are affected.
I also noticed that an old 4.0 build does not have this problem at all.

Any thougt?

dhalbert, 2021-02-08:
This may be a duplicate of #3986, which was fixed by #4122. Could you try the "Absolute newest " build for your board or update from upstream for your custom build?

gc-elec, 2021-02-08:
It worked :)

Thank you very much, problem solved 👍

closed 2021-02-08 by gc-elec