adafruit/circuitpython #4164: RP2040: USB Serial connection sometimes lost

timonsku, 2021-02-09:
I have a hard time reproducing this but I get occasional drop outs for the REPL that are only recoverable after resetting the Pico.
Most of the times this would happen after a code change but also happened just while printing values to the console. It seems to have started to appear after I added the pwmout module but can't say for certain as I haven't extensively used the REPL either before adding that module.
The MSD is working fine after that and code changes are still being applied after the serial connection died. Closing the connection and re-opening does not work, I have to fully reset the chip to be able to see the REPL again.

This is under Windows 10, running CPY nightly `en_US-20210209-5ba28e9` and using Putty as the serial terminal.

dhalbert, 2021-02-09:
Are you using pin GP15? There is a fix needed when using that pin that can interfere with USB in various ways. Try the "Absolute Newest" build from the board page in circuitpython.org.

timonsku, 2021-02-09:
I avoided using that pin.
I was running the build from 3 commits ago, it doesn't seem the last two commits touched anything relevant.

dhalbert, 2021-02-09:
Thanks for the feedback. We'll leave this open, and if you figure out a reproducible case, of course let us know. We recently improved the way we are servicing USB interrupts but there could still be an issue.

jerryneedell, 2021-02-09:
I have been seeing this as well. 
It seems to happen more frequently when I had a button attached to the RUN pin, but it has also occurred with it removed.

timonsku, 2021-02-09:
What OS are you using @jerryneedell ?

jerryneedell, 2021-02-09:
Linux  Ubuntu 20.04

timonsku, 2021-02-09:
Ok that makes sense, I'm pretty sure the need for a reset is a Windows specific issue. Its very easy on Windows to get a COM port into a zombie state where no application can claim it any more unless its removed. I have never seen that on Linux.

jerryneedell, 2021-02-09:
I've also a seen it on a Raspberry Pi.

jerryneedell, 2021-02-10:
FWIW -- on the Raspberry Pi, I have MicroPython on the Pico and I also see occasional spontaneous resets of the board.

hathach, 2021-02-18:
this could be related to https://github.com/hathach/tinyusb/pull/655 , may be worth doing tinyusb update. rp2040 port is still early.

dhalbert, 2021-03-31:
@timonsku #4517, which was just merged, _might_ have something to do with this. If you could try the Absolute Newest build (or rc.0 when it comes out), we would be grateful. Also we have updated TinyUSB a few times which might have already helped. Thanks.

timonsku, 2021-03-31:
I don't do much atm with the RP2040 but when I get back to it I'll make sure to update and let you know if I see it again.

dhalbert, 2021-07-15:
Closing this because there have been many updates to TinyUSB. If you still see this problem, please reopen.

closed 2021-07-15 by dhalbert

Informaticore, 2023-09-05:
Hey - I am not using circuitpython but I have the same issue with micropython AND even with plain Arduino Blinky Sketch. 
If I am running Micropython it was working perfectly fine for like 2 weeks and then this issue appeared. I factory-resetted the device multiple times and tried re-soldering USB connections and everything. I know it is not the right place but maybe someone can help anyway. Here are some facts:

- Device worked perfectly fine for ~2 weeks
- REPL started to disappear
- after boot it worked and after like ~10s the REPL died and I was unable to reconnect to it
- after nuke flash same happens with Arduino Serial Monitor and the Blinky Sketch
- I am not using GPIO15
- it is a custom board (I checked USB Impedance and it should be good)