adafruit/circuitpython #4166: RP2040: hangs when using gamepad

Neradoc, 2021-02-09:
From a problem signaled by Redat on discord, the following code hangs after a reload.
```py
import board
import gamepad
from digitalio import DigitalInOut

button1 = DigitalInOut(board.GP7)
button2 = DigitalInOut(board.GP14)
button3 = DigitalInOut(board.GP13)
button4 = DigitalInOut(board.GP12)
button5 = DigitalInOut(board.GP11)
button6 = DigitalInOut(board.GP10)
button7 = DigitalInOut(board.GP9)
button8 = DigitalInOut(board.GP8)

pad = gamepad.GamePad(
	button1,
	button2,
 	button3,
	button4,
 	button5,
	button6,
 	button7,
 	button8,
)

print("Fin")
```
Save to code.py (or hard reset with the file in place) : it runs once.
Save again (add a comment or whatever) : it hangs (most of the time), the REPL and drive are non responding, reset is required.
Reloading with ctrl-D from the REPL does it too, but the hang happens without needing to be connected to the REPL.
Tested on macOS Catalina.

This code seems to cause the problem every time, but using less pins in GamePad seems to make it happen less reliably.
Note: I'm on latest and it's not using GP15.

tannewt, 2021-02-10:
I suspect this may have something to do with tick management internally. It's a bit of a hack now that I mean to take another look at.

kevinjwalters, 2021-03-10:
I'm getting this too on Pi Pico. Serial console over USB becomes non-responsive, `CIRCUITPY` stops working, etc. I can see the pull-ups do take the pins high. I've got `6.2.0-beta.3-11-g7970c882a on 2021-03-04` on my board.

Are there any possible workarounds for this?

kevinjwalters, 2021-03-10:
BTW, this is currently documented as being limited to eight buttons. Is that correct on all platforms? https://circuitpython.readthedocs.io/en/6.1.x/shared-bindings/gamepad/index.html

dhalbert, 2021-03-10:
> BTW, this is currently documented as being limited to eight buttons. Is that correct on all platforms? 

Yes, that's in `shared-bindings/gamepad/Gamepad.c`. The documentation is a little bit misleading, it's limited to up to 8 buttons, but the `__init__()` can take fewer.

kevinjwalters, 2021-03-10:
Might be worth extending if it's easy as `GamePad` is useful for non gamepad things, I'm using it to get away from crude polling. Its debounce functionality is useful and ability to spot momentary presses while the main program is doing other things. I'm just using it for Maker Pi Pico's three buttons (to GND) at the moment.

dhalbert, 2021-03-10:
We talked about this recently internally, and would certainly entertain a PR to increase the number of supported `DigitalInOut`'s. In addition, the `Gamepad` name can be confused with HID `Gamepad`, which is a different thing. @deshipu, this may be of interest to you.

kevinjwalters, 2021-03-10:
Yes, I advertised name clash as part of: [Adafruit Forums: GamePad vs adafruit_debouncer](https://forums.adafruit.com/viewtopic.php?f=60&t=169041&hilit=not+confused) discussion.

deshipu, 2021-03-10:
I originally wanted to call the module "buttons" (#295), but I guess that was too generic. I'm open to suggestions for another name.

Another option is to introduce a completely new module that would actually handle buttons correctly, using pin change interrupts instead of polling, like the ``gamepad`` does, and then phase out ``gamepad`` — that is however still a little bit beyond my skill at the moment. I suppose the ``gamepad_shift`` would need to stay how it is, though, as the shift registers used in pygamer and pybadge don't have an interrupt pin.

dglaude, 2021-03-10:
I think I have also hit that problem recently with the Adafruit Joy Bonnet and the Anavi Play pHAT:
https://github.com/dglaude/circuitpython_phat_on_pico/blob/main/play_phat.py
https://github.com/dglaude/circuitpython_phat_on_pico/blob/main/joy_bonnet.py

I did not report something but did not link to gamepad, but it is very likely that I was using the gamepad on those two board when I noticed:
https://discord.com/channels/327254708534116352/327298996332658690/817152222143053835

So I believe is that if you run some gamepad code on the Pico, then stop it with CTRL+C the try this from the REPL:
```
import board
import digitalio

```

You have the problem.

tannewt, 2021-07-15:
Closing since we're moving from `gamepad` to `keypad`.

closed 2021-07-15 by tannewt