adafruit/circuitpython #4180: Raspberry Pi-Pico UART not Supported

zack-cpp, 2021-02-11:
Trying to use UART on pi pico board. I started by finding pin names from _[board](https://circuitpython.readthedocs.io/en/latest/shared-bindings/board/index.html)_ modules.
[image]

Based on the _[example](https://learn.adafruit.com/circuitpython-essentials/circuitpython-uart-serial)_ from adafruit, i adjust the TX and RX to GP8 and GP5.

**_`uart = busio.UART(board.GP4, board.GP5, baudrate=9600)`_**

when i run the code, i got this error message:
[image]

I change to the default TX and RX Pi-Pico's pin, referenced to [this closed issue](https://github.com/adafruit/circuitpython/issues/4121#issue-800418879), and still got same error.

is UART for Pi-Pico circuitpython not supported yet?

ladyada, 2021-02-11:
correct, dupklicate of
https://github.com/adafruit/circuitpython/issues/4036
you can sub to that issue to be notified when resolved

closed 2021-02-11 by ladyada

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Once visit this channel for Raspberry Pi Pico Projects

dhalbert, 2021-02-13:
@panchalnikunj It is inappropriate to spam our repos with pointers to your YouTube channel, even if it is relevant material. I am deleting your comments. Please do no do this again.