adafruit/circuitpython #4181: Strange sound Using audioio.AudioOut(board.A0) when in a loop on a Feather M4 Express

riffnshred, 2021-02-11:
Hi guys, I tried to get help on the discord channel but no dice so I'm opening and issue.

I'm playing with the feather M4 express and a stemma speaker. There is something weird happening where after playing the mp3 file, there is a weird noise playing. like a high note going to a low note. kind of line a laser blaster sound FX

Here's a clip of that sound. https://streamable.com/8axwjf

And Here is the code I'm using. 

```
import time
import board
import digitalio
import audiomp3
import audioio

while True:
    try:
        with audiomp3.MP3Decoder(open("whistle.mp3", "rb")) as whistlefx:
            with audioio.AudioOut(board.A1) as a:
                print("playing")
                a.play(whistlefx)
                while a.playing:
                  pass
                a.stop()
                print("stopped")
                time.sleep(10)
    except KeyboardInterrupt:
        break
```

I tried multiple things so far following multiple of the tutorial that uses sound FX on the adafruit website and no dice so far.

Thanks in advance.

ladyada, 2021-02-11:
its due to the capacitors on the audio board going to ground after you deinit audioio. dont use `with` for creating AudioOut - do that at the top of the project so that it doesnt deinit.

closed 2021-02-11 by ladyada

riffnshred, 2021-02-11:
Thanks for the quick response.

Here is my original code and I just tested it again and it still doing the same thing. How do I prevent the deinit in this case? I fooled around a lot last night trying to figure what I'm doing wrong.

```
import time
import board
import digitalio
import audiomp3
import audioio

whistlefx = audiomp3.MP3Decoder(open("whistle.mp3", "rb"))
a = audioio.AudioOut(board.A1)

while True:
    try:
        print("playing")
        a.play(whistlefx)
        while a.playing:
          pass
        print("stopped")
        time.sleep(10)
    except KeyboardInterrupt:
        break

```

Thanks for the help.
Cheers

ladyada, 2021-02-11:
you may have to either mute the speaker if it has a mute, or play a 'loop' of 0 values to the audio stream to keep the output going

riffnshred, 2021-02-11:
gotcha, I'll try this out. Thanks !

riffnshred, 2021-02-11:
Got it !!! adding deinit() after the clip is done prevent it. Thanks!

riffnshred, 2021-02-12:
maybe this info would be also useful. 

what Im working on requires a loop that waits for a request through I2C and depending on what it recieves it will play a specific sound FX.  But the issue was that the AudioOut was initialized but it would not stay up or active while waiting for something. So what I just do is I created a function that initialise AudioOut, play the sound FX and deinit() once the clip is done playing.

here's the function I made instead of trying to keep the audio stream going.
```
def playAudio(clip):
    with AudioOut(board.A1) as a:
        print("playing")
        a.play(clip)
        while a.playing:
          pass
        print("stopped")
        a.deinit()
```