adafruit/circuitpython #4182: ESP-IDF Update to unblock Mac M1 Builds

askpatrickw, 2021-02-11:
When installing the esp-idf from github and runing `source ~/esp/esp-idf/export.sh` it works but in a new terminal when running the export from `ports/esp32s2/` I get `ERROR: Platform Darwin-arm64 appears to be unsupported`.

I can build successfully using the latest IDF (commit 73db14240).

The fix in the IDF appears to be this one
https://github.com/espressif/esp-idf/commit/5739127412b6756801a38c700d068dd1d937ed41#diff-873391a19109de0fec9486d3ef26704a32fe81560a38f2e1f6c4f3db99d19350

Looks like it is time to update the IDF or possibly just pickup this one change.

microDev1, 2021-02-12:
I suggest switching to [`v4.3`](https://github.com/espressif/esp-idf/tree/release/v4.3) branch. The v4.3 [release](https://github.com/espressif/esp-idf/releases/tag/v4.3-beta1) is currently beta.

tannewt, 2021-02-12:
I'm ok if we switch to the 4.3 branch. I'm not planning on doing it myself. I'd expect them to backport this to 4.2 as well.

askpatrickw, 2021-02-13:
I started a PR and did the boring parts of updating the include paths. I don't know C so if someone is excited and could pick up from here, that would be great.

wildestpixel, 2021-04-10:
Im unable to build CP at all on M1 due to the inability to build may-cross from source in setting up my toolchain. Looked at the MicroPython solutions to this but not able to implement on the CircuitPython git sync - I know this detracts that the Espressif nature of this specific issue, but all the same stuck with this atm

dhalbert, 2021-04-10:
If you can't build mpy-cross, then other things will not build as well. What is the error?

wildestpixel, 2021-04-10:
its as follows in a clean case sensitive APFS case sensitive sparse image :

`pixel@Andrews-Mac-mini circuitpython % make -C mpy-cross
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
../py/persistentcode.c:419:2: error: mp_raw_code_save_file not implemented for this platform
#error mp_raw_code_save_file not implemented for this platform
 ^
1 error generated.
Command '['clang', '-E', '-DNO_QSTR', '-Ibuild/tmp', '-I.', '-I..', '-Ibuild', '-Wall', '-Werror', '-Wpointer-arith', '-Wuninitialized', '-std=gnu99', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-Os', '-fdata-sections', '-ffunction-sections', '-fno-asynchronous-unwind-tables', '../py/persistentcode.c']' returned non-zero exit status 1.
QSTR updated
../py/persistentcode.c:419:2: error: mp_raw_code_save_file not implemented for this platform
#error mp_raw_code_save_file not implemented for this platform
 ^
1 error generated.
make: *** [build/py/persistentcode.o] Error 1`

dhalbert, 2021-04-10:
@wildestpixel Try changing line 399 in `circuitpython/py/persistentcode.c` to this:
```c
#if defined(__i386__) || defined(__x86_64__) || defined(__unix__) || (defined(__APPLE__) && defined(__MACH__))
```

wildestpixel, 2021-04-10:
Thats a big step in the right direction I just need to add a path export - may-cross is built, just the arm-eabi to link up :

`/bin/sh: arm-none-eabi-gcc: command not found`

wildestpixel, 2021-04-10:
ah - now at this : 
`pixel@Andrews-Mac-mini atmel-samd % make BOARD=circuitplayground_express 
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
../../py/gc.c: In function 'gc_sweep':
../../py/gc.c:298:68: error: 'MP_QSTR___del__' undeclared (first use in this function)
  298 |                         mp_load_method_maybe(MP_OBJ_FROM_PTR(obj), MP_QSTR___del__, dest);
      |                                                                    ^~~~~~~~~~~~~~~
../../py/gc.c:298:68: note: each undeclared identifier is reported only once for each function it appears in
make: *** [build-circuitplayground_express/py/gc.o] Error 1
`

dhalbert, 2021-04-10:
Do a `make clean BOARD=circuitplayground_express`. The first failure may have messed up the QSTR generation.

wildestpixel, 2021-04-11:
all good I couldn't see the wood for the tress last night before going to bed. I'm building on M1 - Thanks Dan!!!!

closed 2021-06-25 by tannewt