adafruit/circuitpython #4190: USB data stopping

DynamoBen, 2021-02-12:
I created a test program that loops turning a MIDI note on and off. After running for a while (minutes not hours) I lose USB communications (both USB-MIDI and serial debug). I have the LED flashing to debug and it appears the loop is continuing it's just USB comms that stops inexplicably.

This is only reproducible on one of my Wins10 PCs (Dell XPS desktop) and the USB driver is still active when data isn't flowing. Further if I switch to a Teensy 4.0 running the same test (written in C++) USB comms continues indefinitely. 

See forum thread for troubleshooting details: https://forums.adafruit.com/viewtopic.php?f=60&t=175112&p=854548

hathach, 2021-02-13:
There is tinyusb PR that fix this issue recently, please try to manually update tinyusb to see if that fix your issue

https://github.com/hathach/tinyusb/pull/618

dhalbert, 2021-02-13:
@DynamoBen Here is a build for the Pico that updates tinyusb to the tip of its `master` branch as of now. The commit hash is in the filename:
[pico-with-tinyusb-at-09868434c.uf2.zip](https://github.com/adafruit/circuitpython/files/5975194/pico-with-tinyusb-at-09868434c.uf2.zip)

Could you test with this and see if it improves things? Thanks.

I only tested this enough to see that it presents the REPL and CIRCUITPY.

DynamoBen, 2021-02-13:
@dhalbert Some progress, now my test app is locking up instead of USB comms stopping.

DynamoBen, 2021-02-23:
Grabbed latest (adafruit-circuitpython-raspberry_pi_pico-en_US-20210223-cca6cfe) this morning and haven't see any problems. I'll check it again in a few hours and will post an update if it fails.

DynamoBen, 2021-02-23:
Sadly USB comms are still stopping. I will continue to try future builds I guess.

hathach, 2021-02-24:
did you call midi read from your firmware application. The data is stored in midi fifo and need to be retrieved else it will be full and prevent further data transfer. It is mentioned in the hathach/tinyusb#618 discussion

DynamoBen, 2021-02-24:
My tests thus far have been limited to transmitting MIDI. However I did update my test code to include a read but it made no difference.

tannewt, 2021-02-24:
Note we have a TinyUSB update in #4245

DynamoBen, 2021-03-01:
I've been trying the latest build the last several days and the symptom now is the application not running ("locked up").

hathach, 2021-03-02:
can you post your updated script to your first post, so that other could reproduce as well when they try to take a look

DynamoBen, 2021-03-02:
`import time
import board
import usb_midi
import adafruit_midi
from adafruit_midi.note_off import NoteOff
from adafruit_midi.note_on import NoteOn
from digitalio import DigitalInOut, Direction

# MIDI settings
midi = adafruit_midi.MIDI(midi_in=usb_midi.ports[0],
                          midi_out=usb_midi.ports[1],
                          in_channel=(1),
                          out_channel=0)

# Debug LED
led = DigitalInOut(board.LED)
led.direction = Direction.OUTPUT

while True:
    msg_in = midi.receive()
    print(msg_in)

    led.value = True
    midi.send(NoteOn(44, 99))
    time.sleep(0.5)

    led.value = False
    midi.send(NoteOff(44, 0))
    time.sleep(0.5)`

DynamoBen, 2021-03-02:
Test code:

[code.zip](https://github.com/adafruit/circuitpython/files/6069657/code.zip)

hathach, 2021-03-04:
There is several update for rp2040 lately. I will try to update & test with this next. Though I am using Linux, not sure if it has any differences.

dhalbert, 2021-03-31:
@DynamoBen #4517 (now merged, so you can try "Absolute Newest") might be yet another reason for this problem, so if you wouldn't mind testing again, that would be great.

DynamoBen, 2021-04-01:
Grabbed latest just now, no change in behavior. Runs for a little while and then stops.

hathach, 2021-04-02:
@DynamoBen I make an PR #4538 that should fix this issue, I have run your code on pico for an hour, and it still blink as normal. also attached the uf2.zip here [firmware.zip](https://github.com/adafruit/circuitpython/files/6249194/firmware.zip) for quick testing.

DynamoBen, 2021-04-02:
@hathach it does keep blinking, but now I'm back to the original problem. The code continues to run but MIDI stops being sent.

hathach, 2021-04-02:
I am still seeing message fine with my pico, tested on ubuntu 20.04

[image]

DynamoBen, 2021-04-02:
@hathach OK but I'm on Win10

hathach, 2021-04-02:
Thanks for testing, seem like this is another the issue than the one is fixed in the #4538. Will reopen this

hathach, 2021-06-20:
Update: I have fixed another issue with MIDI, https://github.com/hathach/tinyusb/pull/906
though I am not sure if this would fix your issue. I couldn't reproduce it at all. It isn't part of circuitpython yet, you need to manually change the tinyusb to my upstream https://github.com/hathach/tinyusb and manually compile it for testing.

dhalbert, 2021-07-15:
There has been yet another MIDI update in the past couple of days. @DynamoBen Could you retry with an "Absolute Newest" build?

DynamoBen, 2021-07-22:
I'm still seeing the original issue where the code continues to run but MIDI stops in 15-20 mins. I'm seeing this on a Dell XPS 8930 desktop running win10, but I also see it on a Dell Inspiron 5593 laptop running win10 (haven't tested on one previously). So the problem is a bit more widespread than I thought. I've run the exact same test on a custom built non-Dell system and MIDI continued for 3 days w/o stopping.

hathach, 2021-07-22:
while running, did the laptop goes into standby or sleep mode and waked up ?

DynamoBen, 2021-07-22:
> 
> 
> while running, did the laptop goes into standby or sleep mode and waked up ?

No system was plugged in and all power-save settings were disabled.

hathach, 2021-07-22:
> > while running, did the laptop goes into standby or sleep mode and waked up ?
> 
> No system was plugged in and all power-save settings were disabled.

Since I couldn't reproduce it, I will try to modify the code to ouput usb log to uart for you to test with. Will upload it here when done.

hathach, 2021-07-26:
@DynamoBen I have enabled the tinyusb log output to UART @115200 baudrate. It print out all the usb information with level 2 and level 3. Would you mind using this for testing on your device, then post your log as attached filed here for analysis (since it print out quite a lots specially msc communicate very frequent). 
[pico-debug-level.zip](https://github.com/adafruit/circuitpython/files/6875999/pico-debug-level.zip)

DynamoBen, 2021-07-26:
@hathach Using the level2 build and putty (for serial comms) all I see "None." When MIDI stops so does serial data and and I see nothing except for "None" prior to it stopping. Using the level3 build I get a unknown USB device driver error "Device Descriptor Request Failed" so I can't connect.

hathach, 2021-07-26:
hmm that is strange, how would you connect with UART on pico, it should be connect to GPIO0 for UART TX with an ftdi/cp201x. Make sure you have a GND for them, and set flow control to off. Maybe posting the picture of your setup here to see if I could spot any issues with it.

DynamoBen, 2021-07-26:
@hathach Sorry I was pulling serial data from the pico not GPIO0. I was able to capture the log data when MIDI stops, this appears to be the interesting part: 

`USBD Xfer Complete on EP 85 with 4 bytes
  MIDI xfer callback
  Queue EP 85 with 4 bytes ...
USBD Xfer Complete on EP 85 with 4 bytes
  MIDI xfer callback
USBD Unplugged
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Unplugged
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 0E 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 00 09 01 00 00 00 00 00
  Set Configuration
  Open EP 81 with Size = 64
dcd_edpt_open 81
                  Open EP 02 with Size = 64
dcd_edpt_open 02
                  Open EP 82 with Size = 64
dcd_edpt_open 82
                  Queue EP 02 with 64 bytes ...
WARN: starting new transfer on already active ep 2 out
                                                        CDC opened
  Open EP 83 with Size = 64
dcd_edpt_open 83
                  Open EP 03 with Size = 64
dcd_edpt_open 03
                  Queue EP 03 with 31 bytes ...
WARN: starting new transfer on already active ep 3 out
                                                        MSC opened
  Open EP 84 with Size = 64
dcd_edpt_open 84
                  Open EP 04 with Size = 64
dcd_edpt_open 04
                  Queue EP 04 with 64 bytes ...
WARN: starting new transfer on already active ep 4 out
                                                        HID opened
  Open EP 05 with Size = 64
dcd_edpt_open 05
                  Open EP 85 with Size = 64
dcd_edpt_open 85
                  Queue EP 05 with 64 bytes ...
WARN: starting new transfer on already active ep 5 out
                                                        MIDI opened
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes
USBD Xfer Complete on EP 03 with 31 bytes
  MSC xfer callback
  SCSI Command: Test Unit Ready
  Queue EP 83 with 13 bytes ...
USBD Xfer Complete on EP 83 with 13 bytes
  MSC xfer callback
  SCSI Status: 0
  Queue EP 03 with 31 bytes ...
USBD Xfer Complete on EP 03 with 31 bytes
  MSC xfer callback
  SCSI Command: Test Unit Ready
  Queue EP 83 with 13 bytes ...`

dhalbert, 2021-07-29:
@hathach I have just noticed USB disconnects randomly while trying to debug some audio playback on RP2040. I am not doing MIDI. USB becomes disconnected, but when I reconnect, everything is fine (no crash). I will try to narrow this down later. Perhaps the USB interrupt handling is being starved or disabled due to some other operations. I am doing some DMA (audio and display).

hathach, 2021-08-02:
@DynamoBen sorry for late response, looking at your log, look like rp2040 got suspended after awhile. since currently tinyusb port of rp2040 doesn't distinguish between disconnected vs suspend (doable with a bit of extra code with decent testing). Therefore it will cause an re-enumeration. Unfortunately, the dcd isn't clean up all the previous pending transfer which cause the `WARN: starting new transfer on already active ep 4 out`. This can probably cause the failed to transfer as you experience. Suspend detection can take a while, but I think we could address the pending transfer clean up to at least get it running. I will do more test and try to set up something similar to test with. Will post an update of uf2 for testing when complete  

from rp2040 4.1.2.6.4. datasheet
[image]

hathach, 2021-08-02:
> @hathach I have just noticed USB disconnects randomly while trying to debug some audio playback on RP2040. I am not doing MIDI. USB becomes disconnected, but when I reconnect, everything is fine (no crash). I will try to narrow this down later. Perhaps the USB interrupt handling is being starved or disabled due to some other operations. I am doing some DMA (audio and display), though at the time of the disconnect, there is no audio playing, and the display is not changing.

It could probably an different issue, could you provide detail to reproduce the issue, I will try to see if I could figure out something.

dhalbert, 2021-08-11:
@hathach I can try testing with your UART debug output build on several different Windows machines. Would you want a Beagle trace if it does hang up?

hathach, 2021-08-11:
@dhalbert A beagle trace would be great, since it capture what actually happens. UART log does show how the stacks interpret the bus event. The more info the better

PS: I am also trying to troubleshoot this, and will compile an new debug build for testing as well.

hathach, 2021-08-12:
@DynamoBen @dhalbert I tried my best to update tinyusb to address this issue. However, since all my PC cannot reproduce the issue. Would you mind trying this out again. zip contains uf2 for for pico and macropad with multiple debug level (0, 2, 3) . 0 is no debug just for verifying (in case debug log accidentally resolve it by slowing down cpu)
Note: pico uart tx is GPIO0, macropad, uart tx is GPIO20 (SDA in qtstemma connector). Looking forward to your feedback.

[midi-usb-connection.zip](https://github.com/adafruit/circuitpython/files/6974368/midi-usb-connection.zip)

dhalbert, 2021-08-12:
I tried to reproduce on a Dell Optiplex 3020 last night with Windows 10 and failed. I will try an Intel machine with an older motherboard.

Do you have any idea what this might be related to: is it Windows driver related, or might it be more USB controller related? Do you know if USB2 vs USB3 might make a difference? I can try various permutations, but I have many choices right now

hathach, 2021-08-12:
> I tried to reproduce on a Dell Optiplex 3020 last night with Windows 10 and failed. I will try an Intel machine with an older motherboard.

By failed, you mean you couldn't reproduce the issue and everything works well right !?

> Do you have any idea what this might be related to: is it Windows driver related, or might it be more USB controller related? Do you know if USB2 vs USB3 might make a difference? I can try various permutations, but I have many choices right now

I couldn't reproduce the issue myself, and just update code where I think may be the issue. It is hard to tell for sure, If I am right, it is kind of race condition when host issue a bus reset e.g reset duration etc .. and cause the stack (with rp2040 port) to lock up. Theoretically, there shouldn't be any differences between using usb3 or usb2 host, though we all know it is hardly true in practical, there is always some diff, which cause trigger issue especially race condition.

DynamoBen, 2021-08-12:
> @DynamoBen @dhalbert I tried my best to update tinyusb to address this issue. However, since all my PC cannot reproduce the issue. Would you mind trying this out again. zip contains uf2 for for pico [...]

I'm still seeing the issue, most interesting parts below. I used Debug level 2, because level 3 caused the USB driver to not be recognized.

`USBD Xfer Complete on EP 85 with 4 bytes
  MIDI xfer callback
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 17 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 00 09 01 00 00 00 00 00
  Set Configuration
  Open EP 81 with Size = 64
  Alloced 64 bytes at offset 0x180 (0x50100180)
  Open EP 02 with Size = 64
  Alloced 128 bytes at offset 0x1c0 (0x501001c0)
  Open EP 82 with Size = 64
  Alloced 128 bytes at offset 0x240 (0x50100240)
  CDC opened
  Open EP 83 with Size = 64
  Alloced 128 bytes at offset 0x2c0 (0x501002c0)
  Open EP 03 with Size = 64
  Alloced 128 bytes at offset 0x340 (0x50100340)
  Queue EP 03 with 31 bytes ...
  MSC opened
  Open EP 84 with Size = 64
  Alloced 64 bytes at offset 0x3c0 (0x501003c0)
  Open EP 04 with Size = 64
  Alloced 64 bytes at offset 0x400 (0x50100400)
  Queue EP 04 with 64 bytes ...
  HID opened
  Open EP 05 with Size = 64
  Alloced 128 bytes at offset 0x440 (0x50100440)
  Open EP 85 with Size = 64
  Alloced 128 bytes at offset 0x4c0 (0x501004c0)
  MIDI opened
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes
USBD Xfer Complete on EP 03 with 31 bytes
  MSC xfer callback
  SCSI Command: Test Unit Ready
  Queue EP 83 with 13 bytes ...
USBD Xfer Complete on EP 83 with 13 bytes
  MSC xfer callback
  SCSI Status: 0
  Queue EP 03 with 31 bytes ...
  QUSBD Xfer Complete on EP 03 with 31 bytes
  MSC xfer callback
  SCSI Command: Read Capacity10
  Queue EP 83 with 8 bytes ...
USBD Xfer Complete on EP 83 with 8 bytes
  MSC xfer callback
  SCSI Data
  Queue EP 83 with 13 bytes ...
`

hathach, 2021-08-13:
@DynamoBen the usb log look pretty normal, I should add timestamp to the log (will do it next), do you know how long does it take until the  `USBD Bus Reset : Full Speed` occurs. 

1. Does the usb comm both msc and terminal is both freeze ? 
2. Can you still connect / disconnect terminal to cdc
3. Is pico still recognized by windows (even msc/cdc not responding) , you could check it using the usbtreeview software https://www.uwe-sieber.de/usbtreeview_e.html#download 

Look like the issue is different than what I thought it is. Somehow the windows issue bus reset, and it is not related to the physical bus like suspend/disconnect. Probably windows driver think there is an issue with cpy and try to recover by reset it, however, the issue still exist (to windows), and it probably gave up afterwards. Maybe looking at host syslog would help, though I am not sure how to do equivalent  `dmesg` on windows.

DynamoBen, 2021-08-13:
@hathach It usually fails within 20 min, I can get some more exact timing if needed. Terminal continues after MIDI has stopped. I didn't try reconnecting the terminal. Pico is recognized still and no driver errors. 

It's been mentioned by others but I can use other USB MIDI devices without failure and I can run similar test code with a Teensy with no failure. It doesn't help with this specific issue but does mean there isn't a systemic USB issue with newer Dell computers (which seems to be the common thread).

hathach, 2021-08-14:
@DynamoBen device log look normal to me, I think we should look more at the host log/app. Can you tell me more about what does not work when the issue occurs. Original post said both MIDI and serial not working. Can you re-test it.
1. Check if REPL still work
2. Check if MSC still work
3. What is your MIDI  app for testing, maybe either try to close and relaunch it (probably a bus reset cause it to loose connection and could;t reconnect with midi port)
4. Try to test with other MIDI app if possible.

tannewt, 2021-08-31:
Moving this to post-7.0.0. @DynamoBen we'd like your feedback on https://github.com/adafruit/circuitpython/issues/4190#issuecomment-898844231

LibertyBeta, 2021-08-31:
> like your feedback on #4190 (comment)

Will the current patch from tinyusb still be streamed into 7.0.0?
-- 
Rainer Paskiewicz

tannewt, 2021-08-31:
I am hoping to merge #5268 for the next release.

DynamoBen, 2021-09-09:
I'm testing with latest right now, so far it's looking better. I should know tomorrow if the issues I've been seeing are resolved. I plan to do extended testing on two different Dell systems which previously exhibited the failure.

DynamoBen, 2021-09-12:
Been running continuously for days, not sure what changed but things seem to be working now.

hathach, 2021-09-12:
@DynamoBen thank you for your feedback, lots of usb driver updated recently to make it compliance to USB test suite https://github.com/hathach/tinyusb/issues/1059 , which help to review several corner cases bugs. It probably fixes this one as well (hopefully). When you are confident  that the bug is fixed, let us know so that we could close this one.

PR #4538 (2021-04-02): update tinyusb to fix midi buffer overflow issue
update tinyusb to fix #4190 #2473, should wait for user confirmation before merging.

See also https://github.com/hathach/tinyusb/pull/766
PR #5172 (2021-08-18): update tinyusb
Fixes #5125.
Fixes #4190.
May fix some other issues.
closed 2021-09-13 by tannewt