adafruit/circuitpython #4191: usb_midi Device Name

1st-BrainStormer, 2021-02-12:
Device name should be "CircuitPython Midi" not  "CircuitPython Audio" as this is a midi interface.
It would also be useful if this could be overridden with a custom device name.

ToxicTiger901, 2022-05-29:
Sorry I know I'm a bit late to this conversation, but I thought I would add this in case it helps anyone else. 
It's really easy to change the device name if you want to.  Just download the Circuit Python .UF2 file and using a Hex file editor (I used PSPad) search for "CircuitPython Audio".  You can then edit the Text, save the changes and install your modified version of the .UF2 file.
The Pi Pico should then appear with the new name you gave it.

ToxicTiger901, 2022-07-19:
It should still work.
1)	Download CircuitPython (I’m currently using 7.3.1)
[image]

 
2)	Open the .UF2 with a HEX editor (I use PSPAD which is a free text editor) 

3)	Go to line 151100 (or search for “CircuitPython Audio”):
[image]

4)	Change the text for example to “MIDI Pedalboard”.  I haven’t experimented but my guess is that you will be limited to 19 characters.  Fill any unused characters with 00Hex :
[image]
 

5)	Save the changes to the .UF2 file

6)	Copy the file to the Raspberry Pi Pico as you would normally to install CircuitPyton

ToxicTiger901, 2022-07-19:
> I also wanna change the midi device name but it seems like this approach doesn't work anymore. Any tips?

I've added some more instructions. Hopefully they should help.  Let me know if you get stuck!

jwktje, 2022-07-19:
@ToxicTiger901 Tried again and it worked. Thanks a bunch for the detailed update! Still would love a way to just define a name in the Python code, but I don't know if that is even possible without help from Adafruit. Maybe something worth to consider @ladyada ? Totally understand if it's too hard

tannewt, 2022-07-25:
Overall USB names will be configurable with: https://github.com/adafruit/circuitpython/pull/6247 (Testing would be helpful.)

You may also be able to provide a new USB descriptor for the MIDI with: https://learn.adafruit.com/customizing-usb-devices-in-circuitpython

ToxicTiger901, 2023-01-03:
> Overall USB names will be configurable with: #6247 (Testing would be helpful.)
> 

Is this now included in 8.0.0 Beta 6?
If so how do I get it to work?

ToxicTiger901, 2023-01-04:
Thanks.  I think I understand now.

ToxicTiger901, 2023-01-04:
My only slight confusion here is when I configure a Midi Device using standard CircuitPython firmware (see my example Python code above) it comes up as **CircuitPyton Audio**,not **CircuitPython MIDI**?

mrbbp, 2023-09-23:
soory for my newb question.

Is there a way to rename the usb_midi device name?
by default it named CircuitPython usb_midi.ports[x]

i tried to modifing the .uf2 file with an hex editor, but does not work.
i tried with a different length name and the uf2 is not working (no boot)
i tried a second time with the same number of char (13), the uf2 seems to work but the device has the same name `CircuitPython....`

i tried supervisor.set_usb_identification("Herve Bi", "Herve Bi") but no more result.
i'm not sure about my syntax

any help appreciate
regards

circuitPython : 8.2.6
board: waveshare zero

dhalbert, 2023-09-23:
@mrbbp Which name exactly do you want to change? What is the current name and what do you want to change it to?

mrbbp, 2023-09-24:
@dhalbert
i build a small box with 2 rotatives pot on it and a rp2040 to send midi datas.
I use the pot to have a physical interface to manupulate things in Processing.org
 in  my sketch the device listed is `CircuitPython usb_midi.ports[0]`
```
Available MIDI Devices:

----------Input----------  
[0] "Real Time Sequencer"  
[1] "CircuitPython usb_midi.ports[0]"  
----------Output----------   
[0] "Gervill"  
[1] "Real Time Sequencer"  
[2] "CircuitPython usb_midi.ports[0]"
```
i would like to replace this name with a custom name, the name of the box.

regards

dhalbert, 2023-09-24:
Right now that name is not settable from CircuitPython, but you can make a [custom build](https://learn.adafruit.com/building-circuitpython) and change the name. The names that are fixed at compile time are here:
https://github.com/adafruit/circuitpython/blob/1c0155c99b757a1c7840dd6cf6486b1aedfb99d1/shared-module/usb_midi/__init__.c#L182-L185.

mrbbp, 2023-09-24:
hello, i retried with a new rp2040 and the custom (renamed) `.uf2`. Now the device show the new name. I retstart my mac and plug the old rp2040 with custom device but it don't show the good name.

dhalbert, 2023-09-24:
You may need to somehow "forget" the device on the Mac. macOS may have cached information about the device. Have you tried rebooting the Mac?

dhalbert, 2024-03-03:
#8989, which allows renaming the HID interface, is a potential model for code to do this for MIDI.

dhalbert, 2024-04-06:
Fixed by #9146.

closed 2024-04-06 by dhalbert

Antinote, 2025-04-08:
As mentioned above the Mac seems to cache the old name. To forget the old name on the Mac I seem to be able to:

1. Run "Audio MIDI Setup"
2. If the Pico is connected and powered on I can see it in Audio MIDI Setup
3. Disconnect the Pico (You can't delete the device in Audio MIDI Setup unless you disconnect it)
4. Select the Pico in Audio MIDI Setup
5. **Delete the Pico**
6. Reconnect the Pico and it will reappear in Audio MIDI Setup

I called usb_midi.set_names() from boot.py per the instructions on https://docs.circuitpython.org/en/latest/shared-bindings/usb_midi/ . 

I had to use supervisor.set_usb_identification() in boot.py to change the USB Device name of the Pico per these instructions https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.set_usb_identification