adafruit/circuitpython #4194: HC-SR04 + audioio Problem

lawrenceyun90, 2021-02-12:
I am testing a small script where two .wav files are played depending on the readings from an HC-SR04. I am running this script on a feather M4 Express with a Prop-Maker for audio.

In Example 1, I initialize the HC-SR04 before initializing the speaker, and audio will play correctly. In Example 2, I initialize the speaker before the HC-SR04, and the audio comes out as static noise. I should also mention that Example 1 is still broken in a way because the distance readings are off. The readings are proportional (less distance results in smaller reading, more distance results in bigger reading), but they are 5-15x larger than expected. If I delete the speaker-related lines of code, I can get accurate readings from the HC-SR04, so it is not a hardware problem.

Example 1:
```
import time
import math
import board
import digitalio
import audioio
from audiocore import WaveFile
import adafruit_hcsr04

# === hcsr04 ===
pin_hcsr04_trig = board.RX
pin_hcsr04_echo = board.D4
hcsr04 = adafruit_hcsr04.HCSR04(trigger_pin = pin_hcsr04_trig, echo_pin = pin_hcsr04_echo)

# === speaker ===
propMakerPower = digitalio.DigitalInOut(board.D10)
propMakerPower.direction = digitalio.Direction.OUTPUT
propMakerPower.value = True
speaker = audioio.AudioOut(board.A0)
wave = [WaveFile(open("pew1.wav", "rb")), WaveFile(open("pew2.wav", "rb"))]

# === loop ===
while True:
    time.sleep(1)
    #get distance reading
    try:
        #distance in mm
        temp = int(hcsr04.distance*10)
        dist = temp
    except:
        #-1 if sensor fails
        dist = -1
    print(dist)
    
    #play audio
    if dist != -1:
        if dist > 1000:
            speaker.play(wave[0])
        else:
            speaker.play(wave[1])
```
Example 2:
```
import time
import math
import board
import digitalio
import audioio
from audiocore import WaveFile
import adafruit_hcsr04

# === speaker ===
propMakerPower = digitalio.DigitalInOut(board.D10)
propMakerPower.direction = digitalio.Direction.OUTPUT
propMakerPower.value = True
speaker = audioio.AudioOut(board.A0)
wave = [WaveFile(open("pew1.wav", "rb")), WaveFile(open("pew2.wav", "rb"))]

# === hcsr04 ===
pin_hcsr04_trig = board.RX
pin_hcsr04_echo = board.D4
hcsr04 = adafruit_hcsr04.HCSR04(trigger_pin = pin_hcsr04_trig, echo_pin = pin_hcsr04_echo)

# === loop ===
while True:
    time.sleep(1)
    #get distance reading
    try:
        #distance in mm
        temp = int(hcsr04.distance*10)
        dist = temp
    except:
        #-1 if sensor fails
        dist = -1
    print(dist)
    
    #play audio
    if dist != -1:
        if dist > 1000:
            speaker.play(wave[0])
        else:
            speaker.play(wave[1])
```

lawrenceyun90, 2021-02-15:
Just wanted to follow up. I found a temporary solution. Essentially, the HC-SR04 and speaker are constantly initialized and deinitialized within the loop. The two are never initialized at the same time. 

`
import time
import math
import board
import digitalio
import audioio
from audiocore import WaveFile
import adafruit_hcsr04

# === speaker ===
propMakerPower = digitalio.DigitalInOut(board.D10)
propMakerPower.direction = digitalio.Direction.OUTPUT
propMakerPower.value = True
wave = [WaveFile(open("pew1.wav", "rb")), WaveFile(open("pew2.wav", "rb"))]

# === hcsr04 ===
pin_hcsr04_trig = board.RX
pin_hcsr04_echo = board.D4

# === loop ===
while True:
    #get distance reading
    hcsr04 = adafruit_hcsr04.HCSR04(trigger_pin = pin_hcsr04_trig, echo_pin = pin_hcsr04_echo)
    try:
        #distance in mm
        temp = int(hcsr04.distance*10)
        dist = temp
    except:
        #-1 if sensor fails
        dist = -1
    hcsr04.deinit()
    print(dist)
    
    #play audio
    if dist != -1:
        speaker = audioio.AudioOut(board.A0)
        if dist > 1000:
            speaker.play(wave[0])
        else:
            speaker.play(wave[1])
        time.sleep(1)
        speaker.deinit()
`

This is the first time I have reported a bug. When can I expect a fix?

tannewt, 2021-02-16:
> This is my first time reporting a bug. When can I expect a fix?

Unknown. This isn't a priority for Adafruit-funded folks at the moment. There are many other things we're working on now.