adafruit/circuitpython #4196: os.urandom for Pico/RP2040 not supported

joelburton, 2021-02-13:
Attempts to use `os.urandom(n)` on the Pico currently report:

```
>>> os.urandom(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: No hardware random available
```

The datasheet for the Pico indicate that the ring oscillator can be used for this (RP2040 datasheet, page 240). They do note that there are potential cryptographic attacks possible on this, but could this still be a considered "random hardware"? There are CircuitPython libraries that require `os.urandom()`.

ladyada, 2021-02-13:
which library requires os.urandom?

jepler, 2021-02-13:
In the bundle, `adafruit_rsa` uses os.urandom.  In the community bundle, `circuitpython_nrf24l01` does.  In Learn, `CPB_Quick_Draw_Duo` (gracefully works around it if it's missing) and `CLUE_Rock_Paper_Scissors` (advanced version only) do.

ladyada, 2021-02-13:
@joelburton you cant use random.randint() ?

joelburton, 2021-02-14:
I can and can fork those (it’s RSA lib I want). But thought it would be helpful to get this on someone’s non-urgent list for implementation for the pico. 

Thanks!

ladyada, 2021-02-14:
well tbf i dont know the crypto implications - i am not sure how we seed the pico's RNG. i will leave open but yah maybe can you submit a PR to change RSA lib to use random. instead of os.urandom if it throws an exception?

what are you doing with the pico and RSA anyways? makin a U2F? :)

joelburton, 2021-02-14:
I teach Computer Science; I'm making graduation gifts for my students of a
pico with various CS challenges built in, and one of them is decrypting a
message using RSA. :-)

ladyada, 2021-02-14:
ahh ok, random.* would be fine for your purposes then! if you submit a PR we will review it and merge

tannewt, 2021-02-17:
> well tbf i dont know the crypto implications - i am not sure how we seed the pico's RNG. i will leave open but yah maybe can you submit a PR to change RSA lib to use random. instead of os.urandom if it throws an exception?

`random` will use the time you first call it to seed the pseudo-random number generation if `urandom` isn't provided. It should *NOT* be used for cryptography. This true in CPython as well: https://docs.python.org/3/library/random.html Anything that doesn't do cryptography *should* use `random` instead of `os.urandom` because it provides a way to seed the random in cases where we'd want to have fixed behavior for testing.

CPython's [`secrets` module](https://docs.python.org/3/library/secrets.html#module-secrets) is the right higher level random for cryptography and also the reason we should stop using `secrets.py` for storing secret information. (Filed #4212 to track it.)

PR #4213 (2021-02-18): raspberrypi: implement os.urandom
Since the datasheet cast some doubt on the strength of the "rosc_hw->randombit", I use the SHA256 hash function to create a high quality random seed from random values of uncertain entropy, as well as to generate a sequence of random values from that seed using SHA256 as a cryptographically-secure random number generator.

[image]

In practice, it produces over 100kB/s of random data which does not have any gross problems according to _PractRand_ (analyzing limited amounts of data, about 16 megabytes).  Most uses for urandom in CircuitPython (across the Adafru [... cut]
closed 2021-02-18 by tannewt