adafruit/circuitpython #4197: OnDiskBitmap fails quietly when file has been closed

hwiguna, 2021-02-13:
This might be obvious to experienced Python programmers, but not to me.
The code runs with no errors, but the display does not show the bitmap because the with statement has closed the file.
To help beginners like me, it would be good if OnDiskBitmap would throw an exception if it cannot read the file instead of failing quietly as it does now.

```
with open("hari/BlobAnim.bmp", "rb") as bitmap_file:
    bitmap = displayio.OnDiskBitmap(bitmap_file) # setup the file as bitmap data source
print("bitmap ok",bitmap) # this confirms that we have an OnDiskBitmap object.

#... much later in code... using this bitmap does not work because the with open() had closed the file :-(
tileGrid = displayio.TileGrid(bitmap, pixel_shader=displayio.ColorConverter(), width=1, height=1, tile_width=7, tile_height=7)
print("tileGrid ok",tileGrid)
```
    
code.py output:
bitmap ok <OnDiskBitmap>
tileGrid ok <TileGrid>

tannewt, 2021-02-17:
I totally agree we could do this better. I think our best option is to take in the filename instead of a file object. That would imply that OnDiskBitmap manages when the file is open.

There aren't great places to throw an error when a used file is closed because display updates happen in the background. In this example, we could have TileGrid raise an exception but you could always put it in the `with` statement as well and then expect it to continue to work.

So, I think taking in the filename is the better approach. Not sure why I did it this way originally.

hwiguna, 2021-02-17:
Ah the plot thickens. It's probably best to leave it alone rather than changing parameters and breaking existing users' code. Thank you for the explanation.

tannewt, 2021-02-17:
We can definitely evolve it in this direction. It just takes time. In 6.x or 7.x we can add support for string inputs but leave the taking a file object in. Then in 7 or 8 we can remove the backwards compatible support.

jepler, 2021-07-19:
I'll grab this one since I'm already working in OnDiskBitmap today.

PR #5017 (2021-07-19): OnDiskBitmap improvements
 * Fix out of memory write in constructor
 * Allow constructing from filename, rather than from open file object, to improve usability

Closes: #5008
Closes: #4197
closed 2021-07-19 by dhalbert