adafruit/circuitpython #4199: Consistent way of accessing buttons, display and audio on different gaming devices

deshipu, 2021-02-14:
There is a number of gaming devices available that run CircuitPython, and so when a new game is made, it is desirable to make it run on all of them, or at least as many as possible. However, each device has different way of accessing the buttons and the audio, and some builds might even need to have the display initialized.

Asking the players to edit the code of the game to adjust it to the device they have is not a very nice way of doing this. Providing a separate version of the game for each of the devices works better from the player's point of view, but requires a lot of initial work and further maintenance from the creator of the game.

It would be therefore nice if we could have some sort of a game controls library, which would handle the buttons, audio and display for us.

deshipu, 2021-02-14:
One way I attempted to solve this problem with the Stage library is by using a device-specific module — I named it `ugame` because the first device I supported was the µGame, but later I realized that it has to have the same name on all devices. That module contains all the code necessary to initialize and use the buttons, audio and display, and it's frozen on the device. This way each device can come with its own version of the code, specially adjusted to it.

deshipu, 2021-02-14:
Another way of handling this was proposed by @tannewt — instead of having different versions of the library for different devices, there would be only one version, that would contain the code for all the supported devices, and it would guess what device it is running on by checking what names are defined in the `board` module, and initialize it all accordingly.

I tried to implement this here: https://github.com/deshipu/circuitpython-vacuum-invaders-displayio/blob/master/invaders/controls.py

As I see it, there are several downsides to this approach: it takes a lot of memory for defining all the classes that are not used, it makes imports that are not global, and it will give unpredictable results on any new devices that might happen to have similar names defined in their `board` module.

deshipu, 2021-02-14:
Yet another way of doing this would be to generate the library for the individual devices from some kind of a template, letting the player select how the buttons, audio and display are connected in their particular device. This has the advantage of not wasting additional memory for classes that are not being used, and of supporting a wide variety of homebrew devices that we don't need to know up front. It has the disadvantage of requiring the player to select the device they have.

deshipu, 2021-02-14:
Any further ideas are welcome.

tannewt, 2021-02-17:
Thanks for making this issue!

I believe `board` is the right way to approach this problem. It's meant to hold board-specific initialization in a cross-board way. Generally, there should be a standard name for an object that has an expected API. Currently most attributes are pins but we also have I2C, SPI, UART and DISPLAY already.

Let's take these one-by-one:

## Buttons
This looks like we need a way define standard Gamepad objects with a standard pressed mask OR we need an async way of monitoring individual single buttons. This is further complicated by boards that use a joystick for directional buttons. I think this is the biggest question of the three.

## Audio
I think this is simpler because we should have `board.AUDIO` that is an AudioOut or other output depending on the board. Perhaps we need an audio output manager to handle mute and perhaps switching outputs.

## Display
We have `board.DISPLAY` and from your example it seems sufficient.

With these special board names, the game code itself only needs to handle the presence or absence of these things. We could add dummy objects but I'm not sure that's a good idea. It hides the fact that something isn't supported.

jepler, 2021-12-02:
Can we add something like `board.KEYS()` which would return a singleton of one of the keypad classes?

jepler, 2021-12-02:
.. that wouldn't handle the difference between an analog and a digital stick, though.  analog sticks wouldn't work with this scheme.

dhalbert, 2021-12-02:
> Can we add something like `board.KEYS()` which would return a singleton of one of the keypad classes?

The number and position of keys would still vary per device, so I'm not sure this would help making cross-device code that much.