adafruit/circuitpython #4201: Sorting displayio.Group

deshipu, 2021-02-14:
I'm working on a game that uses an isometric view, and one thing I need for this is to sort the sprites representing mobs according to their depth on the screen. In ``displayio`` the drawing order is determined by the order of elements in the ``displayio.Group``, and I'm wondering what would be the best way of keeping it sorted:

1. Create a list out of the group, sort it, and then create a new group, with all the items from the sorted list appended to it. Do this every time the positions of sprites have changed.
2. Perform bubble sort, or similar simple sorting algorithm on the group directly.
3. Create a new kind of group in C, called ``DepthGroup``, and add a ``depth`` attribute to everything that can be added to a group. Automatically track changes to that attribute and sort the group as needed before displaying. Anything that wraps those items can have its own code to update ``depth`` as needed.
4. Add a ``sort`` method to ``Group`` that would behave as on any other Python collection.
5. Some other way I didn't think about.

I wonder what are your thoughts about this? I want to try working on this, though if we choose option 4, I will probably need some help.

deshipu, 2021-02-14:
Something like this should work, but of course not optimally:
```python

for layer in sorted(group):
    group.remove(layer)
    group.append(layer)
```

tannewt, 2021-02-16:
We discussed this in this week's meeting: https://youtu.be/I4uqbDT59Us?t=4446

My suggestion was to move Group to using a proper list for holding its items.

deshipu, 2021-02-21:
Working on this in https://github.com/adafruit/circuitpython/pull/4233

closed 2021-04-02 by deshipu