adafruit/circuitpython #4202: Problems with adafruit_hid not found

gadjet, 2021-02-14:
Hi I've just tried the example code for the keyboard emulation on the Pico. I've loaded CircuitPython 6.2.0-beta.2 onto the Pico and tried a keyboard example "examples/hid_keyboard_shortcuts.py" from Circuitpython.org and I get an error
```
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
ImportError: no module named 'adafruit_hid'
```
Is there a problem in the latest beta as I've seen a Youtube video using this module with 6.2.0=beta.1.

dhalbert, 2021-02-15:
`adafruit_hid` is an external library you can get from the bundle: https://circuitpython.org/libraries. Did you add it to the board?

For general info about libraries: https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-libraries

gadjet, 2021-02-15:
Ah! I didn't realise that, I'm new to Python. I must have missed that in the video or it was mentioned in an earlier video I hadn't watched, thanks for the information I can fix it now.

closed 2021-02-15 by gadjet

dhalbert, 2021-02-15:
Which video were you watching? Thanks.

gadjet, 2021-02-15:
It was by Novaspirit tech on Youtube.

ryanpearsonaudio, 2022-09-21:
I am having a similar error but have the mentioned library in the lib folder on my Pico. I've also verified the _init_ file is an .mpy and not .py. I saw that was an issue elsewhere.