adafruit/circuitpython #4204: RPi Pico i2c.scan() not returning address

h3retek, 2021-02-15:
Similar to this issue:
https://github.com/adafruit/circuitpython/issues/3894

Running: `Adafruit CircuitPython 6.2.0-beta.2-14-g766e79ad3 on 2021-02-12`
Sensor: `mlx90614`

I have this code running:
```
i2c = io.I2C(scl=board.GP5, sda=board.GP4, frequency=100000)
time.sleep(1.0)

# Lock the I2C device before we try to scan
while not i2c.try_lock():
    pass
# Print the addresses found once
print("I2C addresses found:", [hex(device_address) for device_address in i2c.scan()])
 
# Unlock I2C now that we're done scanning.
i2c.unlock()
````

just prints
```
[ ]
```

Although, the sensor is working fine with this added
```
mlx = adafruit_mlx90614.MLX90614(i2c)

while True:
   # temperature results in celsius
   print("Ambient Temperature: ", mlx.ambient_temperature)
   print("Object Temperature: ", mlx.object_temperature)
   time.sleep(1.0)
```

Had it running on `rp2-pico-20210209-unstable-v1.14-11-gc7aaee2b2.uf2`: this prints the address, but the library for the sensor doesn't work (OSError: 5... for whatever unknown reason...).

tannewt, 2021-02-16:
What breakout are you using? Does it include pull ups?

h3retek, 2021-02-17:
[Breakout Used](https://www.instructables.com/Infrared-Temperature-Sensor-MLX90614/)

ladyada, 2021-02-17:
there are no pullups visible on it, pull up resistors (2K - 10K) are required

h3retek, 2021-02-17:
Apologies:

[image]

tannewt, 2021-02-17:
It's possible that the device doesn't like how we probe the address. We usually do empty writes but the peripheral can't do that.

h3retek, 2021-02-17:
Thanks.

I'll take a look at the MicroPython later and get back to you.

h3retek, 2021-02-18:
This is how MP does it:

[machine_i2c](https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2c.c)

Let me know if there is anything I can do to help...

dhalbert, 2021-02-18:
> This is how MP does it:

Looks like MicroPython `[does a bitbang write transaction](https://github.com/micropython/micropython/blob/d4b45898f58aae66c706ca5f832f11727daef46b/ports/rp2/machine_i2c.c#L124-L42)` when the write is 2 bytes or less. (Interesting that it's 2 bytes, not 1 byte).

dhalbert, 2021-03-12:
Fixed by #4315.

closed 2021-03-12 by dhalbert