adafruit/circuitpython #4208: Raspberry Pi Pico PWMAudioOut playback has intermittent problems on 6.2.0-beta.2 absolute latest

kevinjwalters, 2021-02-16:
This sort of thing will demonstrate the problems. Playback is sometimes fine, sometimes accompanies by nasty noise, and occasionally silent with some minor noise.

```
Adafruit CircuitPython 6.2.0-beta.2-18-g2a467f137 on 2021-02-15; Raspberry Pi Pico with rp2040
>>> import board
>>> from audiopwmio import PWMAudioOut as AudioOut
>>> from audiocore import WaveFile
>>> audio_out = AudioOut(board.GP18, right_channel=board.GP19)
>>> testwav = WaveFile(open("120998-32k.wav", "rb"))  ### 32k 16bit stereo 6 seconds
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.play(testwav, loop=True)
>>> audio_out.stop()
>>> audio_out.play(testwav, loop=True)
>>> audio_out.stop()
>>> audio_out.play(testwav)
>>> audio_out.play(testwav)
>>> audio_out.play(testwav)
>>> audio_out.play(testwav)
>>> audio_out.play(testwav)
>>>
soft reboot
```

Can be seen+heard on [Raspberry Pi Pico PWMAudioOut issue I]([https://www.youtube.com/watch?v=QJbMXhWyzvI) (YouTube).

kevinjwalters, 2021-02-16:
Here's zip with wav file: [120998-32k.zip](https://github.com/adafruit/circuitpython/files/5989204/120998-32k.zip)

 It's converted to 32k and trimmed version of https://freesound.org/people/Thirsk/sounds/120998/

kevinjwalters, 2021-02-17:
I don't know if it's a separate issue but copying a file onto `CIRCUITPY` while a looping sound sample is playing makes it crackle for a few second and then stop.

Restarting interpreter with control-D doesn't seem to allow recovery for this. The output just stays low.

tannewt, 2021-02-18:
I think these are an artifact of the CPU or interrupts being blocked so the audio isn't filled in time.

Try providing a bytearray > 1k in size to WaveFile. It'll use the bytearray to buffer then. The larger it is, the more resilient to scheduling the playback will be. https://circuitpython.readthedocs.io/en/latest/shared-bindings/audiocore/index.html#audiocore.WaveFile

kevinjwalters, 2021-02-19:
Is the code based on nrf code? That has (had?) a flaw: https://github.com/adafruit/circuitpython/issues/3030

tannewt, 2021-02-20:
I started with it originally but then based it on the SAMD code.

Did you try larger buffers?

kevinjwalters, 2021-02-21:
I tried with up to `WaveFile(wav_file, bytearray(2 * 4096))`. It crackles in a different way. Copying small files to `CIRCUITPY` still makes it go haywire.

BTW, I think there's a second issue/big here. I've noticed after interrupting and restarting program sometimes it gets stuck while waiting for the sample to complete playing with the typical code of

```
    while audio_out.playing:
        pass
```

kevinjwalters, 2021-02-21:
I did some audio RawSample experiments on command line and then copied working `code.py` onto board and got this

```
Traceback (most recent call last):
  File "code.py", line 198, in <module>
TypeError: 'float' object does not support 'protocol_audiosample'

Code done running.
```

It's definitely not a float! I copied code again and it worked.

kevinjwalters, 2021-03-02:
I get more consistent crackles if I'm doing PWM led (with dc changes) at the same time as playing (non-looping) sample. I've tried it disconnected from host computer too and it still crackles.

kevinjwalters, 2021-03-04:
I thought I'd try upping the buffers a lot.

```
left = WaveFile(left_file, bytearray(2 * 8192))
right = WaveFile(right_file, bytearray(2 * 8192))
```

Those (16k stereo 16bit) play sequentially with some minor distortion but when some PWM GPIO stuff runs and audio plays the Pi Pico dies and becomes non responsive over USB. PWM output is left running. Never appears to recover. I reset via the RUN pin and it does _not_ do the hard fault handler thing. Do you want that as a separate issue ticket

kevinjwalters, 2021-03-17:
I tried 80k which is bigger than the wav file and CircuitPython becomes non responsive at the same point.

kevinjwalters, 2021-03-22:
This example _hangs the board_ after a few seconds if `AUDIO_DAUGHTERBOARD = True` is changed to `AUDIO_DAUGHTERBOARD = False` to do local wave audio: 

https://github.com/kevinjwalters/circuitpython-examples/blob/master/pico/larson-scanner.py

https://github.com/kevinjwalters/circuitpython-examples/tree/master/audio

Serial console dies, CIRCUITPY no longer works, PWM pins are left PWMing.

ladyada, 2021-06-07:
verifying that in 7.0.0 alpha2 this is still happening

dhalbert, 2021-06-07:
> verifying that in 7.0.0 alpha2 this is still happening

Also alpha.3?

ladyada, 2021-06-08:
er, using `main` so its latest latest :) it just prints out alpha-2

dhalbert, 2021-07-06:
#4958 fixed the hanging, but there is still inconsistent scratchiness and also sometimes nothing is played at all. Still investigating those problems.

dhalbert, 2021-08-12:
This should be fixed by #5079, with the default buffer sizes.

closed 2021-08-12 by dhalbert

kevinjwalters, 2021-08-12:
Just had a play with `7.0.0-alpha.6` with some stereo audio playback on a (Maker) Pi Pico and it sounds good.

It does still have a big pop at the start of each `play()` and that includes a second `play()` executed while the one is playing. I checked my sample in audacity and it's silent at the start (and end).

kevinjwalters, 2021-08-12:
My larson scanner no longer kills the interpreter. I now get this.

```
Adafruit CircuitPython 7.0.0-alpha.6 on 2021-08-12; Raspberry Pi Pico with rp2040
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 268, in <module>
  File "code.py", line 185, in start_sound
RuntimeError: Internal audio buffer too small

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

dhalbert, 2021-08-12:
The pops we are going to address via #5136. Doing it nicely with `PWMAudio` is not easy. I can get rid of some but then when I leave the PWM idling it creates white noise.

What is the "larson scanner"? Is that related to #4431? If there is not a current issue, please open a new one. Thanks.

kevinjwalters, 2021-08-12:
Related but the full code is mentioned in https://github.com/adafruit/circuitpython/issues/4208#issuecomment-804296887

I'd been setting the buffer size to 80k to see what it did. I reduced it and the code now runs but there's a fair bit of popping _during_ as the sample plays which must be some clash between (lots of) PWM for other GPIO and the audio GPIO. I'll put a new ticket in for that - I wasn't sure originally if it would get fixed by other work.